mainsail-crew/mainsail #1789: Automatically versioning config files with git

chrisheib, 2024-02-20:
### Requested feature:

I am sadly missing some kind of automatic backup or versioning functionality. 

It would be awesome to have some kind of git integration that optionally did a commit every time 'save and restart' is hit.

chrisheib, 2024-02-20:
Related: https://github.com/gitwatch/gitwatch

Think I'll set that up manually for the time being.

rackrick, 2024-02-20:
Hey there...
mainsail is just the gui running in your browser and talking with moonraker. This request is something which can't be solved in mainsail. To solve this you need somethin running on your printers host.

meteyou, 2024-02-20:
Oh... rick was faster than me...

Yes. If we integrate something like this, we have to download all files from the host, check for changes in the client browser, and then push it to git. This would be very slow and very error-prone.

So, something like this has to be implemented on the printer host itself (maybe in Moonraker). It can provide the current status and some API commands we can handle/display in Mainsail, but the main part has to be integrated into something different.

closed 2024-02-20 by meteyou