mainsail-crew/mainsail #1792: Support for virtual leds plugin

thetic, 2024-02-25:
### Requested feature:

When using the [Klipper virtual LEDs](https://github.com/MapleLeafMakers/klipper-virtual_leds) plugin, I would like to see the virtual LEDs listed on the web UI like "normal" neopixels.

vertexbz, 2024-03-29:
Can be done in the Klipper extension itself [example](https://github.com/vertexbz/tinypixel/blob/master/extension/dummypixel.py)

meteyou, 2024-03-31:
maybe i can work together with the virtual led maintainer. mainsail itself also has a similar function to split long chains (only in the gui, so it doesn't help with macros). so I could display it identically.

but normally, i only integrate native klipper functions and no "unofficial plugins"...

thetic, 2024-03-31:
more context: https://github.com/Klipper3d/klipper/pull/5788

meteyou, 2024-03-31:
@thetic i had a short discussion with @julianschill to be up-to-date about this topic.

there is already a new PR for this feature in Klipper with all missing parts to be mergeable: https://github.com/Klipper3d/klipper/pull/6446

I will check this PR to be ready for a merge and i would love this feature in Klipper & Mainsail!