mainsail-crew/mainsail #1794: "Hide backup files" option does not hide `.moonraker.conf.bkp`

thetic, 2024-02-28:
### Mainsail Version:

2.10.0

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

[image]

### What did you expect to happen instead?

`.moonraker.conf.bkp` is hidden because it is a backup file.

### How to reproduce this bug?

From "Machine > Config Files" enable both "Hidden files" and "Hide backup files." `.moonraker.conf.bkp` remains in the list of files.

meteyou, 2024-02-28:
Hide backup files are only for Klipper Backup files.

Which tool create this backup file? Because it started with a ., it will be also hidden, when you disable the hidden holes. All files with a . as first chars, are hidden files in Linux.

thetic, 2024-02-28:
moonraker creates these https://github.com/Arksine/moonraker/blob/9813011db2e65f13072b8e31ad9a9364e99e338d/moonraker/confighelper.py#L509

meteyou, 2024-02-28:
thx for the link and the FR. I will add these files to "hide backup files".

PR #1801 (2024-03-08): fix: hide moonraker backups when "Hide backup files" is enabled
## Description

This PR add `*.bkp` files to the `Hide backup files` function in Config Files list.

## Related Tickets & Documents

fixes: #1794

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-03-08 by meteyou