mainsail-crew/mainsail #1797: Status report in the log file, generates large logs

TMTYD, 2024-03-01:
### Mainsail Version:

V2.10

### Browser:

Firefox

### Device:

Other (Please tell us below)

### Operating System:

Linux

### What happened?

When the printer is running, quite large logs are generated because every status is written to the log file. This means that a log file can be around 40mb in size or even larger.

`=============== Log rollover at Fri Mar  1 00:00:00 2024 ===============
Stats 226888.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7270 freq=167996618  ra
spberry_pi: temp=38.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.11 cputime=1200.830 memavail=3670928 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226889.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7271 freq=167996618  ra
spberry_pi: temp=39.4 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.10 cputime=1200.838 memavail=3670576 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226890.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7272 freq=167996618  ra
spberry_pi: temp=39.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.10 cputime=1200.843 memavail=3669552 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226891.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7273 freq=167996618  ra
spberry_pi: temp=40.4 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.10 cputime=1200.849 memavail=3670832 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226892.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7274 freq=167996618  ra
spberry_pi: temp=39.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.10 cputime=1200.854 memavail=3670832 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226893.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7275 freq=167996618  ra
spberry_pi: temp=39.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.10 cputime=1200.859 memavail=3670832 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226894.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7276 freq=167996618  ra
spberry_pi: temp=38.5 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.34 cputime=1200.868 memavail=3670832 print_time=595.833 buffer_time=0.000 pr
int_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226895.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9
bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7277 freq=167996618  raspberry_pi: temp=39.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.34 cputime=1200.879 memavail=3669664 print_time=595.833 buffer_time=0.000 print_stall=0 extruder: target=180 temp=180.0 pwm=0.400
Stats 226896.7: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=45820 bytes_read=135572 bytes_retransmit=9 bytes_invalid=9 send_seq=3581 receive_seq=3581 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=7278 freq=167996618  raspberry_pi: temp=38.9 heater_bed: target=65 temp=65.0 pwm=0.284 sysload=0.34 cputime=1200.889 memavail=3670420 print_time=595.833 buffer_time=0.000 print_stall=0 extruder: target=180 temp=180.0 pwm=0.400`

[image]

### What did you expect to happen instead?

Not every print status is written to the log file, but rather a shortened version

### How to reproduce this bug?

If the printer is off, the log file is not as large because not every printer status is written to the log file

meteyou, 2024-03-01:
Mainsail is just a Webinterface and only runs in your browser. You report something about Klipper. Mainsail can only display it or download it. It has nothing to do with the log itself.

FYI: these huge log files are very important to find issues...

closed 2024-03-01 by meteyou

TMTYD, 2024-03-02:
oh OK then that was probably wrong here, I thought Klipper and mainsail belong together, that may be that you have to use the log files to find out any errors, but that could be classified as debugging or not?

meteyou, 2024-03-02:
Klipper and Mainsail work very closely together, but Mainsail is only executed on the client pc/smartphone and shows you the current status of Klipper (the data is loaded via Moonraker (the API of Klipper)).

you don't know beforehand when you need the debug log, so it's 99% very good that the log looks like this. Klipper also has a python script to filter the "debug" output from the klipper log.

but i have to say that all helpers really need the debug part in the logs to be able to help! so for me personally, leaving it out would be more of a handicap than a help.

TMTYD, 2024-03-02:
That may all be correct and I also understand that the log is important in order to correct any errors. But assuming you only have a 16GB memory card in the pi and some programs may be running on it, which, just as an example, takes up half of the space on the memory card. And then you uploaded some gcode files, e.g. which are 5-10mb in size, which may happen with large models and if you also have a print job of, say, 24 hours, then I think that log will be very, very large. Maybe you could make the above-mentioned output appear if you activated the debug mode, then the log file wouldn't actually be that big.

meteyou, 2024-03-03:
the log rotates every 24h and after 3-5 days (i would have to look into the code. moonraker and klipper have different settings, one 3 days, the other 5 days). so you only have a few of these files saved at any time.

to be honest. you are not the first person to use this system (we currently have about 3000 downloads per day) and the logs are definitely not a problem. at least a minimum 16gb memory card is needed and a minimum 32gb memory card is recommended. especially since full print beds gcodes can have 500mb or more.

dw-0, 2024-03-03:
> But assuming you only have a 16GB memory card in the pi and some programs may be running on it, which, just as an example, takes up half of the space on the memory card

That is mistake #1. Take a dedicated SBC and a well sized SD card. So you pretty much found out your issue yourself. Don't do what you did.

>  And then you uploaded some gcode files, e.g. which are 5-10mb in size, which may happen with large models

5-10mb are nothing compared to what other people upload to their printers. It could go into the gigabytes for some models.

> Maybe you could make the above-mentioned output appear if you activated the debug mode

Which output should Mainsail make appear where?
You were already told that the log is created by Klipper. What appears in Mainsail doesn't matter in terms of logsize. We could show a single character of that logfile or double the displayed content. The filesize on the system won't change. What is your actualy proposal that belongs in the scope of Mainsail? I don't see one...