mainsail-crew/mainsail #1810: Toolhead controls for extra ABC axis in dashboard

naikymen, 2024-03-08:
### Requested feature:

I've been using Mainsail a lot and it is really great. So first thanks a lot for it!

I have modified klipper to control 3 extra axes, and would like to have controls for them in Mainsail's dashbaord.

I'd like to know if it would be simple to (optionally) add controls for these additional axes to the ToolHead control component.

For example, an extra A axis could show up like this:

[image]

### Solves the following problem:

There is also at least one other Klipper fork implementing this feature (from gear), and several multi-axis printers have been spotted in the wild. I am hoping that this is a growing trend.

Machines with more than 3 kinematic axes can't be controlled with Mainsail's dashboard (other than through manual input of GCODE). Adding controls for them would be convenient.

### Additional information:

I'd say this could be really useful to others working on multi-axis printers, or other multi-axis CNC machines in general.

UGS has added features and visualization for ABC axes already, which might serve as a reference: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.12

Unfortunately UGS does not talk to Moonraker (yet): https://github.com/winder/Universal-G-Code-Sender/wiki/Connecting-the-Controller

I hope this is interesting to you!

meteyou, 2024-03-08:
thx for your request. i found here also a gif with the implementation in Universal-G-Code-Sender: https://github.com/winder/Universal-G-Code-Sender/pull/1920

do you have any idea, how to visualize these axis in the other control options?

is this also with generic klipper possible? because i only implement generic klipper options right now and dont support klipper forks.

naikymen, 2024-03-08:
> is this also with generic klipper possible? because i only implement generic klipper options right now and dont support klipper forks.

Generic Klipper does not implement this at all, and given its pace and the experimental stage of multi-axis 3D printing I don't expect it to implement it for the next 5 years at least. FR dismissed!

If it is not too much to ask, I'd like to know how Mainsail would be modified to accomodate this.

The simplest idea I have is to write a new component for the dashboard, with controls for the additional axes. I'd be glad to learn a bit of frontend if it can get me going with this idea.

Thanks for the quick response!

naikymen, 2024-03-08:
> do you have any idea, how to visualize these axis in the other control options?

LinuxCNC states that "_The A B C axes refer to rotational coordinates about the X Y Z axes respectively_".

I'm not a designer but I imagine that they should probably hint at this information.

naikymen, 2024-03-09:
Well it was not as terrible as I thought.

On the Klipper side, this small modification to `motion_report.py` was enough: https://gitlab.com/pipettin-bot/forks/firmware/klipper/-/commit/484037b7684b54b694ca17793a7e65d78f356c1e

On the Mainsail side this worked just fine: https://gitlab.com/pipettin-bot/forks/firmware/mainsail/-/commit/81b4656bdc29b23b54a9c60c469da150d865994c

ChatGPT and a well written source really helped me figure out what copypasta was needed. I'm really happy with this. :)

[image]

A note: the control "steps" and feedrate I have hijacked from the Z axis. That's untidy, but this was just to play around and learn a bit.

closed 2024-03-09 by naikymen

vipulchari, 2024-03-09:
Well this was something I was working on , thanks you got it working

naikymen, 2024-03-10:
> Well this was something I was working on , thanks you got it working

@vipulchari cool! Are you also working on a multi-axis Klipper?