mainsail-crew/mainsail #1818: Vivaldi download issue

Zeanon, 2024-03-14:
### Mainsail Version:

v2.10.0

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I try to download a log, gcode, moonraker db json or any other file from mainsail with Vivaldi, a new tab opens(as expected) but it stays with about:blank in the url bar.
Downloading from mainsail using chrome works and downloading from any other website using Vivaldi also works, so I guess it's some sort of issue between mainsail and Vivaldi.

### What did you expect to happen instead?

Being able to download stuff from mainsail

### How to reproduce this bug?

Install the latest version of Vivaldi and try to download a gcode file from your printer

meteyou, 2024-03-14:
Do you use my.mainsail.xyz?

Zeanon, 2024-03-14:
no, just normal local ip
what I just realized, could it be cause some js stuff gets blocked by one of my extensions?(though it has worked for quite some time, the issue only arose about 2 weeks ago)
Console unfortunately gives no information at all

meteyou, 2024-03-14:
if you use any "js blockers", then we cannot help you...

closed 2024-03-14 by meteyou

Zeanon, 2024-03-14:
I have just checked, there are no js blockers active.
Whats strange is that it worked till about 2 weeks ago.
I even used a completely fresh install of vivaldi to check and it still does not work, so not an extension

meteyou, 2024-03-14:
to be honest. Vivaldi itself has many "blockers". You have to doublecheck it with different browsers.

Mainsail itself just open a new tab with the http download link. so there is no "function" in Mainsail itself for the download. Its "just a link".

Zeanon, 2024-03-14:
ok, fair enough
do you have any idea what to look for?

meteyou, 2024-03-14:
at first, pls double check it with a "normal" browser...

Zeanon, 2024-03-14:
I double checked with chrome and it works
All the chromium based settings are the same between my chrome and vivaldi (to rule them out)

Zeanon, 2024-03-23:
Coming back to this:
After I restarted Vivaldi on my Laptop (without changing any settings or plugins at all) I now can not download files on vivaldi either so I guess it has something to do with a recent vivaldi update

Zeanon, 2024-03-23:
do you by any chance have any idea on how it could be fixed?

meteyou, 2024-03-24:
no sry and we also don't support browser with any blocking features, because every update could be break something. this is just annoying for developing.