mainsail-crew/mainsail #1819: Unable to download log files

sochartgit, 2024-03-16:
### Mainsail Version:

v2.10.0

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Tried on Chromium and Firefox and Firefox ESR

As small files are opening straight in the web page, large ones are intended to be downloaded but it does not work.
Download starts and fails immediately.
Using WGET, I do have number of retries and 206 errors
2024-03-15 20:45:59 (11,5 MB/s) - Connection closed at byte 1911046. Retrying.

--2024-03-15 20:46:09--  (try:17)  https://3d.XXXXXX.XXXX/server/files/logs/telegram.log
Connecting to 3d.caxa.ca (3d.XXXXXX.XXXX)|192.168.13.242|:443... connected.
HTTP request sent, awaiting response... 206 Partial Content
Length: 22794240 (22M), 20883194 (20M) remaining [text/plain]
Saving to: 'telegram.log.1'

### What did you expect to happen instead?

I do expect that the file downloads properly onto my machine

### How to reproduce this bug?

Go to the machine config option and try to download a log file over 50kB

closed 2024-03-16 by github-actions[bot]

sochartgit, 2024-03-16:
Wow,
This is no help request at all. This is a bug.
Thanks for closing this ticket without even trying to understand.

meteyou, 2024-03-16:
@sochartgit Mainsail itself only provide the URL to download the logfile. This URL looks be right, when your browser can start downloading it. The download itself, will only handled by your browser itself. So it couldn't be a bug of Mainsail... For me, it looks like wrong settings on your nginx instance (or your reverse proxy).