mainsail-crew/mainsail #1834: FR: prompt on unsaved probe offset and SAVE_CONFIG

jamincollins, 2024-03-26:
### Requested feature:

I believe it would be nice to have a confirmation dialog if there is an unsaved z offset when **SAVE_CONFIG** is pressed.

### Solves the following problem:

Numerous times I've gone through a first layer calibration and forgotten to save the offset before saving the config.

meteyou, 2024-03-31:
I think you mixed up your "requested feature" and "solves the following problem" descriptions. The SAVE_CONFIG button has nothing to do with "changing files". The SAVE_CONFIG button is at the top bar of Mainsail and will save the z_offset, not your open config file (and this button is not available when you change your config file).

But if I only read your "solves the following problem" description, I can only reject this feature request because it is no longer possible for me to recognize what klipper has changed after SAVE_CONFIG and which are your changes, and this would instead lead to "broken printer.cfg" if I would automatically merge these two config files again.

So, I can only recommend that you click on the SAVE_CONFIG button if you have done a "first layer calibration" before you change your config file yourself.

closed 2024-03-31 by meteyou

jamincollins, 2024-04-01:
It seems I've failed to adequately describe the issue in text.

When printing a first layer calibration and adjusting the **Z-Offset**, I wind up with this UI state:
[image]

As you can see both **SAVE CONFIG** and the Z-Offset **SAVE** are present and enabled.

I'm asking for the UI to prompt/warn when pressing **1** in this case, as doing so (AFAICT) loses the Z-Offset.

meteyou, 2024-04-02:
Thx for your second description. This is an entirely different workflow than I first expected and a valid point.

I also have a dialog on my to-do list when you click the SAVE_CONFIG button to show all "changing settings". I could add this feature to this dialog.