mainsail-crew/mainsail #1838: Ability to Add additional info to the webcam page....lots of empty space that could be useful.

PhilBaz, 2024-03-31:
### Requested feature:

Looking at the webcam page i was suddenly struck by all the empty space and started to think about filling it with some key info. For instance temps and console.

In a sense it could be a much more useful page if it had the ability to add elements from the dashboard to one side, while maintaining the webcam image as the primary element. say dividing the screen vertically and having the webcam as 2/3 of the screen and a single vertical list in 1/3 of the screen.

Lots of times I want to be on the webcam page and have the image be the primary element... by adding one or two more elements (for me temp and console) the webcam page would take on a whole new use/relevance.
 

### Solves the following problem:

 Paging back and forth between webcam page and dashboard. Make webcam page more inviting, useful, relevant.
 

### Additional information:

Hope someone likes this...it could be nice!

Image #1  current webcam page.
[image]

Image #2 UI mockup
[image]

meteyou, 2024-03-31:
No... We don't want a second dashboard and the webcam page should display the webcam as big as possible. in your case, the webcam will be smaller than before.

closed 2024-03-31 by meteyou

PhilBaz, 2024-03-31:
@meteyou  It could be arranged so that the webcam doesn't loose size, or it could go below the webcam.  It just since you can't do anything, and can't see any critical info on the webcam page you can't really stay on there.