mainsail-crew/mainsail #1847: sensor_type: temperature_combined messes up the web UI

RNGIllSkillz, 2024-04-13:
### Mainsail Version:

v2.10.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When configuring temperature_fan with sensor_type: temperature_combined
the web interface will not show fan name also it will not change target temp the dashborw with an error: The value '' is not valid for TEMPERATURE_FAN

### What did you expect to happen instead?

Web interface should show fan name and it should change target temp.

### How to reproduce this bug?

Configure temperature_fan with sensor_type: temperature_combined, go to the dashboard. Name for that fan will be blank and you will not be able to change target temp for that fan.

meteyou, 2024-04-14:
pls upload your klippy.log. this help us to reproduce this issue.

meteyou, 2024-05-04:
I cannot reproduce this issue. I added a "Max_MCU" temperature_fan, but it works like it should:

[image]

closed 2024-05-04 by meteyou