mainsail-crew/mainsail #1851: Replace Stack Overflow logo with material icon

Ashk3000, 2024-04-19:
### Requested feature:

Replace Stack Overflow logo in heightmap tab material icon.

### Solves the following problem:

Logo usage against [Stack Overflow guidlines](https://policies.stackoverflow.co/company/trademark-guidance/)

### Additional information:

[image]

dw-0, 2024-04-19:
@Ashk3000 Please point out the parts of the Stack Overflow guidlines that are violated with the use of the stack overflow (material) icon in Mainsail.

Ashk3000, 2024-04-19:
> @Ashk3000 Please point out the parts of the Stack Overflow guidlines that are violated with the use of the stack overflow (material) icon in Mainsail. 

I added the quote.

dw-0, 2024-04-19:
> > @Ashk3000 Please point out the parts of the Stack Overflow guidlines that are violated with the use of the stack overflow (material) icon in Mainsail.
> 
> I added the quote.

[image]

closed 2024-04-19 by dw-0