mainsail-crew/mainsail #1852: Add a maintenance page

dtjager, 2024-04-23:
### Requested feature:

Include a button that flags hour marker in moonraker database that maintenance was completed. Option to view another timer of "Hours since maintenance" or "Last Maintenance: " kind of thing. Maybe include filament used as well for people who like to run abrasive filament.

Possibly add you can record what maintenance was done, like checkboxes (such as belt tension, grease, input shaper, nozzle change, filter change, etc). Or a notes section for other stuff. 

### Solves the following problem:

Potentially help reduce troubleshooting/maintenance issues. Identify poor components. Ability for users to easily compare longevity of hotends, heaters, belts, etc.

meteyou, 2024-04-23:
This is already done and there exists already another FR...

https://github.com/mainsail-crew/mainsail/pull/1274

closed 2024-04-23 by meteyou

dtjager, 2024-04-23:
I do not have a reminders tab anywhere or a (+) with any printer I have had.

meteyou, 2024-04-26:
Because it is not merged & released until now. but there is already an FR for the same feature, thats why i closed this one, because it is only a duplicate.

dtjager, 2024-04-26:
Fantastic. That will be helpful.