mainsail-crew/mainsail #1864: No switching after adding printer

wfjo852, 2024-04-30:
### Mainsail Version:

v2.11.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

[image]

When a printer is added, monitoring and information are loaded.

[image]
[image]

If you try to call the printer, the connection becomes impossible due to a Moonraker error.

### What did you expect to happen instead?

normal connection

### How to reproduce this bug?

1. Update everything to the latest version using kiauh
2. Register the printer on the network
3. Switch printer

### Additional information:

None

szafran81, 2024-04-30:
[image]

Same here after yesterdays update (before it it worked just fine)

wfjo852, 2024-04-30:
After downgrading mainsail to v2.8.0, it can be used normally.

szafran81, 2024-04-30:
In my case v2.10.0 allready works fine

rackrick, 2024-04-30:
Hey guys...
thanks for the report. I could reproduce the issue. There is indeed a problem with mainsail 2.11 and the farm view.
We fix this soon and report back to you.