mainsail-crew/mainsail #1868: Overlay on video

ariekraakjr, 2024-05-02:
### Is your feature request related to a problem? Please describe

For the alignment of my IDEX toolheads I mounted usb cam under the bed.
for easy alignment of the nozzles is want to use a transparent overlay on my video output.
[image]

### Describe the solution you'd like

create options like:

```
overlay_image: /path/to/overlay.png
overlay_h_align: left|center|right
overlay_v_align: top|center|bottom 
```

### Describe alternatives you've considered

I now use an esp32cam with some own code to do the overlay.
It works, but i like it more integrated on the machine.(no second wifi device)

mryel00, 2024-05-02:
Worth mentioning here is #1557

ariekraakjr, 2024-05-02:
Thanks for the info.

mryel00, 2024-05-02:
I moved the FR to the Mainsail Repo. So I'm gonna reopen it, to keep track of this issue.

filamonkey, 2024-09-03:
I think this would be really useful for IDEX, dual nozzle and tool changers. My proposal would be a simple overlay with styling on the cameras dashboard view and be able to use the toolhead controls to align both nozzles and obtain the offset form their relative positions.
[image]

PR #2053 (2024-12-01): feat(Webcam): add a optional overlay for IDEX calibration
## Description

This PR adds an option to enable a crosshair in the webcam and modify the color and the diameter of the circle. This only adds this option in the most common webcam clients (Mjpgstreamer, mjpgstreamer-adaptive and webrtc-camerastreamer).

## Related Tickets & Documents

fixes #1868 

## Mobile & Desktop Screenshots/Recordings

[image]

## [optional] Are there any post-deployment tasks we need to perform?

Maybe we should add this option also in other webcam clients.
closed 2024-12-01 by meteyou