mainsail-crew/mainsail #1874: Multiple Devices for Power On

mlee12382, 2024-05-04:
### Requested feature:

It would be great if we could select multiple devices in the settings to be controlled by the Power On button in the Mainsail UI. My printers have multiple power relays that need to be turned on in order for the printer to boot and be usable and I'm sure that I am not the only one. I have relays for both 24v and 48v PSUs as well as for the AC mains Bed. Would be nice to be able to use the single Power On button to toggle everything instead of having to switch them individually in the Power Menu.

meteyou, 2024-05-04:
Would it be possible, that you only enable 1 PSU to starting klipper itself and enable the rest with a startup macro?

mlee12382, 2024-05-04:
> Would it be possible, that you only enable 1 PSU to starting klipper itself and enable the rest with a startup macro?

Not in my case no, there is a separate relay to power the board in addition to the main 24v psu relay so that I can power lights or other 24v devices without the printer also being on for maintenance purposes. 
KlipperScreen allows to turn on multiple devices with a single button, it should be possible to change the settings for mainsail to also work this way.