mainsail-crew/mainsail #1875: Adding notes for G-code files

julianCast, 2024-05-05:
### Requested feature:

Add notes to g-code files.
[image]

### Solves the following problem:

It would greatly help to track print results with g-codes. Specially for issues. For example, if we have similar g-codes with different results this would help to quickly differentiate between the pieces, or even adding links to sites.

### Additional information:

I could help with this if it's something people would find useful besides me.

meteyou, 2024-05-05:
Thats why you can make notes in your history. I don't see any benefit in adding notes to "new gcodes".

dw-0, 2024-05-05:
I agree with @meteyou on this. I don't really see the point of linking the note to the gcode file, especially not if it is already possible in the history entry of the corresponding print job. If you delete the gcode file, the note is gone. Also i don't really see the point of keeping bad gcode files that give a bad print result.

julianCast, 2024-05-05:
> Thats why you can make notes in your history. I don't see any benefit in adding notes to "new gcodes".

Oh I didn't know about that feature! That's perfect for the purpose, thank you then and I'll close this

closed 2024-05-05 by julianCast