mainsail-crew/mainsail #1880: Force Move Missing

FoldZero, 2024-05-12:
### Requested feature:

[image]
I have to switch back and forth between Mainsail and Fluid as i need this feature when fixing print crashes. 
I would prefer to stay on Mainsail. 

The feature should unlock the motors and allow the directional control full movement, caveat emptor

### Solves the following problem:

I have to switch back and forth between Mainsail and Fluid as i need this feature when fixing print crashes. 
I would prefer to stay on Mainsail.

meteyou, 2024-05-12:
we decided not to implement it, because you can easily destroy your whole printer with it. if you need it, you are welcome to use a macro for it.

closed 2024-05-12 by meteyou

FoldZero, 2024-09-20:
caveat emptor. 
i strongly disagree with baby sitting, for such a useful feature in the hands of the experienced.
A settings checkbox and a warning?
I can but ask. :D

meteyou, 2024-09-24:
> caveat emptor. i strongly disagree with baby sitting, for such a useful feature in the hands of the experienced. A settings checkbox and a warning? I can but ask. :D

We might implement it once we have worked through the entire backlog, but right now, we have more important features to implement. Maybe a contributor will do it. We are concentrating on essential functions that every user can use to simplify everyday life with Klipper.

I'm also an experienced Klipper user, but I don't know of any use cases in the last few years where I needed `force_move`. This feature is considered way too important and very often misused. For this reason, it's hazardous for users who think they are experienced but are still learning.
Just my 2 cents

TL ;DR
If you think this feature is essential for you, feel free to fork it and if you implement a good solution to protect new users, feel free to create a PR.