mainsail-crew/mainsail #1882: Weird acceleration values by running SHAPER_CALIBRATE (FlyAdxl345 by Mellow)

Surax98, 2024-05-14:
### Mainsail Version:

v2.11.2

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

The SHAPER_CALIBRATE post-processing phase returns very weird data

### What did you expect to happen instead?

Getting correct acceleration values

### How to reproduce this bug?

Run SHAPER_CALIBRATE using a Flying ADXL345 by Mellow

### Additional information:

The following is what I get by running the X axis calibration by SHAPER_CALIBRATE and waiting for the output:
```bash
Recommended shaper_type_x =  zv, shaper_freq_x = 54.2 Hz
To avoid too much smoothing with '3hump_ei', suggested max_accel <= 0 mm/sec^2
Fitted shaper '3hump_ei' frequency = 48.0Hz (vibrations = 11.3%, smoothing ~= 0.497)
To avoid too much smoothing with '2hump_ei', suggested max_accel <= 0 mm/sec^2
Fitted shaper '2hump_ei' frequency = 41.0 Hz (vibrations = 12.3%, smoothing ~= 0.456)
To avoid too much smoothing with 'ei', suggested max_accel <= 0 mm/sec^2
Fitted shaper 'ei' frequency = 50.4 Hz (vibrations = 22.2%, smoothing ~= 0.221)
To avoid too much smoothing with 'mzv', suggested max_accel <= 0 mm/sec^2
Fitted shaper 'mzv' frequency = 33.0 Hz (vibrations = 18.3%, smoothing ~= 0.297)
To avoid too much smoothing with 'zv', suggested max_accel <= 2400 mm/sec^2
Fitted shaper 'zv' frequency = 54.2 Hz (vibrations = 34.8%, smoothing ~= 0.142)
```

This is instead the output by using the same collected CSV data, but manually executing the klipper calibrate script:
```bash
 ~/klipper/scripts/calibrate_shaper.py /tmp/calibration_data_x_20240514_135758.csv -o /tmp/shaper_calibrate_x.png
Fitted shaper 'zv' frequency = 55.4 Hz (vibrations = 35.7%, smoothing ~= 0.057)
To avoid too much smoothing with 'zv', suggested max_accel <= 12000 mm/sec^2
Fitted shaper 'mzv' frequency = 33.2 Hz (vibrations = 18.5%, smoothing ~= 0.185)
To avoid too much smoothing with 'mzv', suggested max_accel <= 3200 mm/sec^2
Fitted shaper 'ei' frequency = 50.4 Hz (vibrations = 22.2%, smoothing ~= 0.127)
To avoid too much smoothing with 'ei', suggested max_accel <= 4700 mm/sec^2
Fitted shaper '2hump_ei' frequency = 43.2 Hz (vibrations = 13.3%, smoothing ~= 0.289)
To avoid too much smoothing with '2hump_ei', suggested max_accel <= 2000 mm/sec^2
Fitted shaper '3hump_ei' frequency = 51.0 Hz (vibrations = 12.4%, smoothing ~= 0.315)
To avoid too much smoothing with '3hump_ei', suggested max_accel <= 1800 mm/sec^2
Recommended shaper is zv @ 55.4 Hz
```

It is pretty obvious that there is something strange going on, either by Mainsail side or Moonraker side (if that's the case, I will open the issue on Moonraker's GitHub page).

meteyou, 2024-05-14:
@Surax98 mainsail is only a webgui. All gcodes will be executed by Klipper itself.

Surax98, 2024-05-14:
> @Surax98 mainsail is only a webgui. All gcodes will be executed by Klipper itself.

Yeah, that was my thought but i can't find any other explanation on why the webgui returns some values while the python scripts other ones. Maybe Moonraker loses some values for whatever reason? Should I open a issue on theri GH?

meteyou, 2024-05-14:
No. Moonraker only pass the data

closed 2024-05-17 by meteyou