mainsail-crew/mainsail #1883: Different printer different web address

Ale11Re, 2024-05-16:
### Requested feature:

I'd like the option to use a different web address for different printer. I.E. if I navigate to mainsail web interface, let's say 10.0.0.2 I can choose the printer I want via selection. The problem is that the printer interface will be available on 10.0.0.2 no matter what printer I choose. It'd be great if choosing the first printer its interface will be available on 10.0.0.2/printer1; choosing the second printer will be on 10.0.0.2/printer2 etc.

### Solves the following problem:

It would eliminate the need to manually choose printer UI via orca by specifying web address path specific to the printer.

meteyou, 2024-05-16:
Just create seperate nginx configs or use the redirect

closed 2024-05-16 by meteyou

Ale11Re, 2024-05-16:
All right, thank you. I'll do my best configuring it.

meteyou, 2024-05-16:
this was the PR for the redirect. i hadnt time until now to write the docs: https://github.com/mainsail-crew/mainsail/pull/1837

feel free to open a post in the support-forum on discord or here in the discussions section, if you need help to setup it.