mainsail-crew/mainsail #1891: Password Required during manual install of Mainsail.

ankonaskiff17, 2024-05-20:
### Mainsail Version:

2.11.0

### Browser:

Other (Please tell us below)

### Device:

Other (Please tell us below)

### Operating System:

Other (Please tell us below)

### What happened?

While working my way through a manual install of Mainsail (NOT MainsailOS) on a RasPi running current version of Pi OS installed via the Pi Imager.
When I got to what I will call the first check, prior to NGINX installation entered the command `http://<printer-ip>:7125/server/info` everything checked good in that I got the big block of information that was expected.
The next step,  Install web server & reverse proxy (NGINX) I worked my way through the process, mostly just copy/pasting the various lines and generating files using Nano.
On the second check by entering `http://<printer-ip>/server/info` I got a 404 error on the web page. Went back through the whole NGINX process line by line in the event I had a typo or a Nano file didn;t get written. Everything was there and in the correct locations. The very last step `sudo systemctl restart nginx` I think is the problem because when I executed that restart nginx, I got this. 

> ==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Authentication is required to restart 'nginx.service'.
Authenticating as: ,,, (pi)
Password:

Once I entered my password, I got the below 

> ==== AUTHENTICATION COMPLETE ====
I then entered `http://<printer-ip>/server/info` and got the big block of information that implies things are set up correctly.

### What did you expect to happen instead?

I was not expecting a 404 error. Was hoping I got server info.

### How to reproduce this bug?

I think if you just follow the process as written for a manual installation it should cause the same problem since there is no mechanism in a copy/paste process to accomodate password entry.

Instead of 

> mkdir ~/mainsail
sudo rm /etc/nginx/sites-enabled/default
sudo ln -s /etc/nginx/sites-available/mainsail /etc/nginx/sites-enabled/
sudo systemctl restart nginx

Instead

> mkdir ~/mainsail
sudo rm /etc/nginx/sites-enabled/default
sudo ln -s /etc/nginx/sites-available/mainsail /etc/nginx/sites-enabled/

And instruct user to then do single line because when I did the entire block I did not get any error or request for password 

> sudo systemctl restart nginx

### Additional information:

I have noticed that PiOS is getting really tight with things requiring user Pi's password compared to past practice.  It's been several months since I have messed with any Pi's and my memory is thinking that using sudo was good enough. In this instance I needed both sudo and my password.

meteyou, 2024-07-14:
sry for the long delay. i tested it right now and i cannot reproduce this issue. i tested it with a new flashed sdcard with raspberry pi os lite (bookworm, 64bit).

maybe you used a "full" version instead of the lite version. but i think this guide should also only show commands and the user should also read the responses. this guide is also only for "linux known users" and not for "linux beginners". there are much better options for beginners available (mainsailOS or KIAUH. both are recommended instead of the manual setup)

closed 2024-07-14 by meteyou

ankonaskiff17, 2024-07-16:
I think this ended up a Raspberry Pi issue rather than a Mainsail problem. It just manifested itself while I was doing Mainsail stuff.

 

 

 

 

Wayne Wells

email: ***@***.***

Cell: (803) 553-6091

 

10682 Two Notch Rd.

Apt. 7107

Elgin, SC 29045

meteyou, 2024-07-16:
thx for your feedback!

ankonaskiff17, 2024-07-30:
As a final addendum, I spent too many hours battling the printer (Ender-3 Pro) and threw in the towel about a month ago and based on the preponderance of "It just works" comments in regards to a Bambu Labs P1S with the four spool feed system.
It was purely the printer, not Klipper or Mainsail.  The software was installed and was working fine so no complaints about the software. 
I bought a printer to make functional prints and to teach myself CAD, not fight with a poorly built, undocumented piece of garbage.
I would caution you that other manufacturers will be chasing that Bambu so they can get that same "It just works" review.  Each person that buys one of those printers is lost to Klipper, Marlin, Octoprint, Fluidd, Mainsail and many other firmwares and front-ends. 
Once the local high schools start back and settle down I'm giving that aggravating Creality printer to the STEM department of my local school district. It's just the type of project those teenagers will dive into.