mainsail-crew/mainsail #1899: Unreliable macro prompt opening

HelgeKeck, 2024-06-06:
### Mainsail Version:

v2.11.2

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Macro prompts does not open always

### What did you expect to happen instead?

that the prompt opens every time i call it

### How to reproduce this bug?

jsut try to open a prompt multiple times

### Additional information:

Youttube video demonstration
https://www.youtube.com/watch?v=C0w-u5VQP-g

Macro that has been used for this test
```
[gcode_macro TEST2]
gcode:
	RESPOND TYPE=command MSG="action:prompt_begin Load Filament T0"
	RESPOND TYPE=command MSG="action:prompt_button PLA...............215°C|_START_LOAD_FILAMENT TEMP=215"
	RESPOND TYPE=command MSG="action:prompt_button PETG..............230°C|_START_LOAD_FILAMENT TEMP=230"
	RESPOND TYPE=command MSG="action:prompt_button ASA...............260°C|_START_LOAD_FILAMENT TEMP=260"
	RESPOND TYPE=command MSG="action:prompt_button ABS...............255°C|_START_LOAD_FILAMENT TEMP=255"
	RESPOND TYPE=command MSG="action:prompt_button PC................275°C|_START_LOAD_FILAMENT TEMP=275"
	RESPOND TYPE=command MSG="action:prompt_button FLEX..............240°C|_START_LOAD_FILAMENT TEMP=240"
	RESPOND TYPE=command MSG="action:prompt_show"
```

HelgeKeck, 2024-06-19:
any comments on this one?

meteyou, 2024-06-19:
@HelgeKeck, I haven't had time to test it yet, but I think it's just an interference of the console lines due to the fast clicking. Macro prompts are very slow, and it's more of a "console hack" instead of a real klipper function. I think you have to live with that.

HelgeKeck, 2024-06-20:
> @HelgeKeck, I haven't had time to test it yet, but I think it's just an interference of the console lines due to the fast clicking. Macro prompts are very slow, and it's more of a "console hack" instead of a real klipper function. I think you have to live with that.

the console commands i used was jsut for reference. even when calling from normal macros or macro buttons it is unfortunately unreliable. so its not guaranteed that a modela will show or not

PR #1918 (2024-06-23): fix(macroPromts): fix internal close function
## Description

This PR fix the computed showDialog in the macro prompts. I switch from array pos to timestamp from the event entry. The macro prompt didn't open, when you opened the same macro prompt multiple times without a refresh before.

## Related Tickets & Documents

fixes: #1899 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-06-23 by meteyou