mainsail-crew/mainsail #1900: Camera FPS css issue

Jaizu, 2024-06-06:
### Mainsail Version:

v2.11.2

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

FPS text is not readable

### What did you expect to happen instead?

Having readable text

### How to reproduce this bug?

Check the camera fps counter

### Additional information:

[image]

Jaizu, 2024-06-06:
solved on my computer with custom css but cant apply that on mobile
`@-moz-document url-prefix("http://192.168.0.17:4409/") {
    .webcamFpsOutput[data-v-72cc9e9c] {
        color: white;
    }
}`

meteyou, 2024-06-06:
Thx for your bugreport! I created a PR to fix this issue, but i used a white background in the light theme, instead of changing the text color.

Jaizu, 2024-06-06:
Whatever you think that fits better ^^

PR #1901 (2024-06-08): fix(webcam): fix fps output in light mode
## Description

This PR fixes the contrast of the fps output in mjpegstreamer and mjpegstreamer-adaptive in the light theme.

## Related Tickets & Documents

fixes: #1900 

## Mobile & Desktop Screenshots/Recordings

after: 
[image]

before:
[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-06-08 by meteyou