mainsail-crew/mainsail #1922: Web UI fails to load on Safari 17.0.0

PMarinov1994, 2024-06-29:
### Mainsail Version:

2.11.2

### Browser:

Safari

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

When I go to the web interface via a web browser that uses webkit2gtk (Safari 17.0.0), the web UI fails to starts and crashes

### What did you expect to happen instead?

The web UI to load normally

### How to reproduce this bug?

I am running EndeavourOS with Arch. To reproduce this bug, the easiest way is to use the BadWolf browser (https://wiki.archlinux.org/title/Badwolf). It is based on webkit2gtk which uses the Safari 17.0.0 engine. This is important because most 3D slicers for linux use webkit2gtk to show the Device tab (OrcaSlicer is my choice and it is where I saw this bug)

The follwoing image shows the log before the crash
[image]

dw-0, 2024-06-29:
Please install a different browser, chromium based for example, or firefox and report back if you could reproduce the issue. Thanks!

PMarinov1994, 2024-06-29:
> Please install a different browser, chromium based for example, or firefox and report back if you could reproduce the issue. Thanks!

I am using Firefox as my main browser and there is **no issue** there. The problem is that many modern slicers have an integrated _Device tab_ which shows the web UI of the printer. Here is OrcaSlicer for example:

[image]

Many of them use integrated browsers. The windows version uses Microsoft Edge which comes from Microsoft's WebView SDK. It works without a problem. But for Linux, I believe they use webkit2gtk most of the time since this is the most widely used  SDK and this is the reason for my report. Other WEB Pages load fine

[image]

So I cannot blame this problem to the slicer's build in browser. If you know of a workaround or where the issue lies, I can try to fix it just for me, but I am not sure why this is happening.

meteyou, 2024-06-29:
the title of this issue is probably written in a very misleading way. mainsail itself works fine on (real) safari itself:

[image]

PMarinov1994, 2024-06-29:
I have edited the title from Safari 17.0.0 to webkit2gtk. Again the problem is on Linux which uses webkit2gtk which is as far as I understand based on Safari. On Mac you are using the native Safari (and its embedded version) which does not have any issues.

dw-0, 2024-06-29:
> So I cannot blame this problem to the slicer's build in browser.

I think we can. As the inbuild browser of OrcaSlicer is currently the only factor that is causing the reported issue.
Mainsail is running fine on Linux in any other browser, running fine on Windows and Mac on any other browser.
Even on Safari, which is powered by WebKit not to be confused with webkit2gtk, which is a Linux port of WebKit.

There are actually some other issues regarding OrcaSlicer and webkit2gtk:
https://github.com/SoftFever/OrcaSlicer/issues?q=sort%3Aupdated-desc+is%3Aissue+is%3Aopen+webkit

This is not a Mainsail issue.

PMarinov1994, 2024-06-29:
> I think we can. As the inbuild browser of OrcaSlicer is currently the only factor that is causing the reported issue. Mainsail is running fine on Linux in any other browser, running fine on Windows and Mac on any other browser.

At first that is what I though too. But this is when I wanted to try a different browser that uses webkit2gkt engine. This is how I found BadWolf browser. The issue is happening there as well. And other web sites work fine in BadWolf. It is only Mainsail as far as I tested that is having problems. Maybe you can point me to other sites that use similar technology as Mainsail for me to test them. If I find another site that fails to load its JavaScript in BadWolf, than the problem is for sure in webkit2gtk engine

PMarinov1994, 2024-06-29:
I also want to say that I have looked through all of the webkit2gtk on Linux related issues in OrcaSlicer's repo and yes there are issues that break webkit from starting, but after fixing them, everything in Orca is working as expected except the Mainsail UI.

meteyou, 2024-06-29:
To be honest, if Mainsail normally works in any normal browser but not in this particular case, which also makes other strange appearances, then the problem is not with Mainsail but with the browser integration in Orcaslicer.

closed 2024-06-29 by meteyou

Noisyfox, 2024-10-30:
This happen in PrusaSlicer, BambuStudio, OrcaSlicer, badwolf browser on Ubuntu 24.04 (and other distos that ships with libwebkit2gtk-4.1) , which all happen to use libwebkit2gtk to display a web view in the app.
Unsure which one to blame, just some extra information if someone found here.

meteyou, 2024-10-30:
@Noisyfox thx for this detail information! Are you able to test it with an older or an newer version from libwebkit2gtk?

broncosis, 2024-11-04:
I can confirm it works with a  older version of libwebkit2gtk because when I updated it it broke my update logs are in the orca
issue https://github.com/SoftFever/OrcaSlicer/issues/7210  there is at least 4 issues on the orca github related to this but I have also confirmed its a issue in Prusa slicer as well I haven't bothered to add my klipper printer into Bambu studio.  I can spin up a vm and test various version if need be   I am also currently running libwebkitgtk-6.0-4 libwebkitgtk-6.0-dev  and that doesn't solve the problem

coryrc, 2025-08-02:
Isn't this an issue in `webkit2gtk` that *just happens* to be triggered my mainsail? Webkit should not be crashing under any circumstance (any crash is a potential exploit).