mainsail-crew/mainsail #1924: Gcode Viewer - Tracking only works with quality set to Medium or lower

NameOfTheDragon, 2024-06-30:
### Mainsail Version:

2.11.12

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

The "Tracking" mode always displays the full print if the quality is set to High, Ultra, or Max. I've been away from 3D printing for a few months but have recently built a new Voron V0.2 and made a fresh install of Mainsail, and was surprised to find that tracking does not appear to work on the higher quality rendering modes.

### What did you expect to happen instead?

I'm pretty sure that tracking mode used to work at all quality settings.

### How to reproduce this bug?

- go into GCode Viewer
- select "Current File"
- Turn on Tracking mode
- Set the rendering quality to High, Ultimate or Max.
- Observe that the fill print is always displayed.

Sindarius, 2024-07-01:
Thanks for reporting this I will take a look, have you checked to see if you have any errors showing up in the browser's console when you try to use a higher render quality?

TopoDiFogna, 2024-07-02:
No errors in console, it happens also on https://sindarius.com/

Example Gcode causing the issue: [Sovol SV06_clothes peg_PLA_0.2_1h33m.zip](https://github.com/user-attachments/files/16064006/Sovol.SV06_clothes.peg_PLA_0.2_1h33m.zip)

Sindarius, 2024-07-02:
I was able to recreate this issue and am working a fix. A later version of BabylonJS changed a default value of a parameter on a call which caused the updating to not function properly.

closed 2024-07-03 by meteyou

NameOfTheDragon, 2024-07-20:
> I was able to recreate this issue and am working a fix. A later version of BabylonJS changed a default value of a parameter on a call which caused the updating to not function properly.

Thanks for looking at this. I can confirm it does work now. However, I noticed that the GCode display window is now blank so your change might have had a knock-on effect there. It's not a major issue for me as I hardly ever want to see the gcode, but I just happened to notice it stopped working some time after I originally reported this issue.

meteyou, 2024-07-20:
@NameOfTheDragon pls post a screenshot of you gcode display window. Maybe it is only a "caching issue". pls try to reload the page with CTRL-SHIFT-F5 (on windows) or CMD-SHIFT-R (on mac).

NameOfTheDragon, 2024-07-20:
I captured this screenshot, then I noticed you suggested Ctrl-Shift-F5, so I switched back to the Mainsail window to try it and, without me even doing anything, it was working! I have no idea what happened there :( I cant make it fail now.

HOWEVER, simulation mode is broken and behaves like a thing posessed, switching back and fore between the finished object, the part-printed object and sometimes anywhere in between. Again, this is not an issue for me because I never really use that mode, but I thought you;d want to know.

[image]

meteyou, 2024-07-20:
@NameOfTheDragon thx for the report! i think i found the issue, but Sindarius has to double-check it.

PR #1926 (2024-07-03): fix(gcodeviewer): update gcodeviewer to fix rendering issues
## Description

This PR update the gcodeviewer to fix rendering issues with high/max quality.

## Related Tickets & Documents

fixes: #1924 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none