mainsail-crew/mainsail #1938: Maintenance reminders popping before expected

idmadj, 2024-07-09:
### Mainsail Version:

2.11.2

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Started using maintenance reminders (really cool feature). I've set up three, some based on meters, others on meters + hours, but the reminders seem to popup in the notification panel way too early, far before the maintenance is due. Initially I thought this was because the maintenance needed to be performed once after creating it, but even after a perform, the reminder came back prematurely.

_[image]

### What did you expect to happen instead?

I'd expect the maintenance reminders to appear in the notification panel when the maintenance is actually due.

### How to reproduce this bug?

Create a maintenance item that's due far in the future or in a lot of meters, then use the printer for a few hours. Eventually the reminder will pop, long before the maintenance is due.

### Additional information:

I've double-checked that the time and timezone is correctly set on the system.

meteyou, 2024-07-09:
Thx for the report! This is already fixed, but we didn't publish a new release until now: https://github.com/mainsail-crew/mainsail/pull/1871

idmadj, 2024-07-09:
> Thx for the report! This is already fixed, but we didn't publish a new release until now: #1871

Wasn't #1871 published with 2.11.2? That's the version I'm using.

meteyou, 2024-07-09:
@idmadj no... When it is released, there is a note at the end of the PR with the release version number.

idmadj, 2024-07-09:
@meteyou OK I was just asking because the release notes for 2.11.2 mention #1871: https://github.com/mainsail-crew/mainsail/releases/tag/v2.11.2

meteyou, 2024-07-09:
ohh... you are right! it looks like this tag isn't showing on a mobile phone. i see it on the mac now. i reopen this issue.

pls post the output of:
```
http://<ip or hostname>/server/history/totals
http://<ip or hostname>/server/database/item?namespace=maintenance
```

idmadj, 2024-07-09:
OK, thanks for looking into it :)

Here are the outputs:

/server/history/totals
```json
{
  "result": {
    "job_totals": {
      "total_jobs": 1320,
      "total_time": 12775254.3934642,
      "total_print_time": 12375846.1676831,
      "total_filament_used": 14024977.2326233,
      "longest_job": 66729.060230653,
      "longest_print": 66328.111771586
    },
    "auxiliary_totals": []
  }
}
```

/server/database/item?namespace=maintenance
```json
{
  "result": {
    "namespace": "maintenance",
    "key": null,
    "value": {
      "494786a5-77ed-4001-b303-bc3ec2a618ee": {
        "name": "Recharge dessicant",
        "note": "",
        "start_time": 1720375109.381,
        "end_time": null,
        "start_filament": 13953426.3605933,
        "end_filament": null,
        "start_printtime": 12314028.0776268,
        "end_printtime": null,
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": false,
            "value": 0
          },
          "printtime": {
            "bool": false,
            "value": 0
          },
          "date": {
            "bool": true,
            "value": 90
          }
        }
      },
      "4ee8a119-8002-4211-8d6f-c70e7bd40a90": {
        "name": "Level bed",
        "note": "",
        "start_time": 1720494497.752,
        "end_time": null,
        "start_filament": 14024977.2326233,
        "end_filament": null,
        "start_printtime": 12375846.1676831,
        "end_printtime": null,
        "last_entry": "dc66dc30-ba03-4c78-a585-11299331882f",
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": true,
            "value": 250
          },
          "printtime": {
            "bool": true,
            "value": 50
          },
          "date": {
            "bool": false,
            "value": 0
          }
        }
      },
      "6b80e150-6284-4c8c-a587-7706e4b5b5d1": {
        "name": "Inspect belts, bearings and screws",
        "note": "",
        "start_time": 1720375024.291,
        "end_time": null,
        "start_filament": 13953426.3605933,
        "end_filament": null,
        "start_printtime": 12314028.0776268,
        "end_printtime": null,
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": true,
            "value": 5000
          },
          "printtime": {
            "bool": false,
            "value": 0
          },
          "date": {
            "bool": false,
            "value": 0
          }
        }
      },
      "966a06d9-2ed7-4558-b04c-2ceb98ecc32f": {
        "name": "Clean & grease leadscrews + oil rods",
        "note": "",
        "start_time": 1720374761.751,
        "end_time": null,
        "start_filament": 13953426.3605933,
        "end_filament": null,
        "start_printtime": 12314028.0776268,
        "end_printtime": null,
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": true,
            "value": 500
          },
          "printtime": {
            "bool": true,
            "value": 100
          },
          "date": {
            "bool": false,
            "value": 0
          }
        },
        "select_id": "maintenance_966a06d9-2ed7-4558-b04c-2ceb98ecc32f"
      },
      "dc66dc30-ba03-4c78-a585-11299331882f": {
        "name": "Level bed",
        "note": "",
        "start_time": 1720374873.072,
        "end_time": 1720494497.752,
        "start_filament": 13953426.3605933,
        "end_filament": 14024977.2326233,
        "start_printtime": 12314028.0776268,
        "end_printtime": 12375846.1676831,
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": true,
            "value": 250
          },
          "printtime": {
            "bool": true,
            "value": 50
          },
          "date": {
            "bool": false,
            "value": 0
          }
        },
        "perform_note": null
      },
      "ee411204-c4cf-417c-a955-00651e86d4b3": {
        "name": "Replace nozzle",
        "note": "Also replace heatblock if cannot screw nozzle enough to contact heatbreak",
        "start_time": 1720374962.968,
        "end_time": null,
        "start_filament": 13953426.3605933,
        "end_filament": null,
        "start_printtime": 12314028.0776268,
        "end_printtime": null,
        "reminder": {
          "type": "repeat",
          "filament": {
            "bool": true,
            "value": 5000
          },
          "printtime": {
            "bool": true,
            "value": 1000
          },
          "date": {
            "bool": false,
            "value": 0
          }
        }
      }
    }
  }
}
```

meteyou, 2024-07-14:
@idmadj thx for the report! i found the issue. it was a wrong converting from units. the total number is in mm and the input in the maintenance feature is in m. so if you print 5000mm, it triggered the 5000m entry. i fixed it and it will be part of the next release.

closed 2024-07-14 by meteyou

idmadj, 2024-07-14:
@meteyou That's awesome, thanks for the quick fix, can't wait to give it a try!

PR #1941 (2024-07-14): fix(maintenance): fix filament trigger for maintenance entries
## Description

This PR fix the filament trigger for maintenance entries. The factor for converting mm to m was missing in the getter.

## Related Tickets & Documents

fixes #1938 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none