mainsail-crew/mainsail #1944: Ability to enter custom gcode in the "Pause at layer" dialog.

miklschmidt, 2024-07-15:
### Requested feature:

Since the macro used for this feature supports custom gcode macro's, it would be ideal if the dialog allowed the user to input a custom macro / gcode command instead of only picking between M600 and PAUSE.

This could either be achived by changing the select box to a combobox/autocomplete which accepts custom text input and suggests M600 / PAUSE by default, or by adding a "Custom" option and showing a text input based on the selection of that item. 

[image]

### Solves the following problem:

- Provides access to existing functionality
- Flexible support for multiple use cases including:
  - Remapping toolheads / filaments at certain layer heights on MMU / IDEX systems after the fact.
  - Turning on/off any aux fans or chamber filter fans
  - Taking pictures
  - Cancelling a print at a certain layer height
  - Calling moonraker agent methods, for example push notifications to come back to the print to watch a tricky part of the process.
  - Reducing or increasing chamber temperature
  etc. etc.

### Additional information:

The idea is to provide ample opportunity for the user to "save a print" if he forgets to add these during slicing, or is printing the same sliced file but needs an escape hatch at a certain layer.

In the future (not a requirement , just extra "icing on the cake" so to speak, the existing functionality already accounts for a vast array of use cases), it could even allow multiple entries, alas that would require changes to the mainsail klipper macros, so i suggest tabling that for now.

miklschmidt, 2024-07-15:
@meteyou done! #1945

XScorpion2, 2024-07-15:
This is Xulkal on the RatRig discord.
I'm available if you need someone to test out these features before landing.
Depending on my schedule, I might try implementing this and putting up a pr for review, but not holding my breath this week due to family matters.

meteyou, 2024-07-15:
hey @XScorpion2!

at the moment i have to finish 2-3 things before i can start with this FR. so i probably won't get to these FR until next week. If you are interested, you are welcome to contribute this FR. i would be very happy about it!