mainsail-crew/mainsail #1955: Driver OTPW cannot be cleared or ignored

miklschmidt, 2024-07-21:
### Mainsail Version:

2.12.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

A lot of buttons that do nothing. This seems unintentional. 
https://github.com/user-attachments/assets/d6d23126-de5c-4748-95f0-57b18351ee6c

It would be nice if they worked. Currently you cannot avoid this warning if you want to get the maximum out of a set of 2209 sticks for example, and as such this is just an annoyance and not very helpful.

### What did you expect to happen instead?

The notification to be dismissed / ignored, or worst case scenario that the buttons were disabled.

### How to reproduce this bug?

Either generate the OTPW flag on a driver, or just unconditionally push it to the notification list in the client code since it's a mainsail-only implementation.

sbtoonz, 2024-08-05:
[image]
appears to still be present in 2.12 release

meteyou, 2024-08-05:
@sbtoonz pls post a video...

meteyou, 2024-08-05:
@sbtoonz pls try a reload without cache (ctrl + shift + f5 on windows)

sbtoonz, 2024-08-05:
force cache reload not yielding any different outcome

meteyou, 2024-08-05:
ohhh... @sbtoonz pls check the PR for this bug itself. this one is not even merged yet (waiting for a review) and therefore not yet released!

sbtoonz, 2024-08-05:
Ah I'll pull this down then :)

meteyou, 2024-08-05:
You can download a build from the ci chain.

PR #1956 (2024-11-07): fix(notifications): fix dismiss function for tmc warnings
## Description

This PR fix the dismiss function for TMC warnings in the notifications.

## Related Tickets & Documents

fixes #1955 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-11-07 by meteyou