mainsail-crew/mainsail #1966: Option to show history chart/table as time instead of number of prints

Gunner087, 2024-08-09:
### Requested feature:

It would be nice to have the ability to display the history chart/table below with print time instead of number of prints. I think the print time display option would be a more accurate representation of the different categories. For example, if you cancel a print on the first layer or before the print even starts because you forgot something, that print would contribute very little to the "cancelled" prints from a time perspective compared to a 12hr print that contributes to "completed". Obviously this is not something very important, but thought I would mention it since it would potentially be more accurate.

[image]

### Additional information:

Most of my coding skills are self taught, but if someone can point me to the right areas of the code, I am willing to learn and try to implement this feature.

meteyou, 2024-08-11:
Thank you very much for your feature request!

When I read your FR for the first time, I thought it would be useless and that just changing the ratio between "error" and "successful" would be different but would have no added value.

Then, I discussed this FR internally with the other crew members and gave it more thought. Now, I see great potential in it, especially when you do the filament-based calculation. How much filament was used successfully, and how much was "wasted". I just have to find a good position for the switch between count, filament, and time-based calculation.

Gunner087, 2024-08-11:
Awesome! The successfully used vs. "wasted" filament information was one of the benefits I had in mind, but failed to specifically mention above. I was also thinking of the time calculation being useful in a similar way. How much time is wasted on cancelled prints vs. just knowing how many prints were cancelled.

PR #2007 (2024-11-09): feat(History): add option to show stats in different values
## Description

This PR adds an option to switch the stats (circle or table) in the history page between amount, filament and time.

## Related Tickets & Documents

fixes #1966 

## Mobile & Desktop Screenshots/Recordings

Circle Amount:
[image]

Circle Filament:
[image]

Circle Time:
[image]

Table Amount:
[image]

Table Filament:
[image]

Table Time:
[image]

## [optional] Are there any post-deployment tasks we need to perform?

none

## Summary by Sourcery

Introduce a feature to toggle the display of history statistics between dif [... cut]
closed 2024-11-09 by meteyou