mainsail-crew/mainsail #1990: feature: show disk usage

heitorPB, 2024-09-04:
### Requested feature:

I'd like to see how much disk space is left in my 3D printer, in addition to memory and CPU usage.

### Solves the following problem:

Right now, I have no way to know how much space is used, how much is left, and the total space on the printer, unless I SSH into it.

rackrick, 2024-09-04:
[image]

Check the g code site for free disk space

heitorPB, 2024-09-04:
Oh, it is in the g-code section.

Can it be added to the Machine page as well? So that we have the full information about the host there?

meteyou, 2024-09-04:
its also in the machine page:
[image]

but pay attention! in the maschine page, you will see the free space from the drive/partition where the config directory is. on the gcode page, you will see the free space from the drive/partition where the gcode path is. on a normal setup, both is on the same partition.