mainsail-crew/mainsail #2001: Adding a new webcam gives a javascript error

Rodin10, 2024-09-08:
### Mainsail Version:

2.12.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I'm trying to add a webcam to Mainsail via the settings. Everything is setup correctly, I enter the name and when I hit save the webcam isn't added and I get the error index-17a5ec1d.js:38 Response Error: Invalid value for parameter 'rotate' (no action) in my Chrome console.

### What did you expect to happen instead?

The webcam to be added and available

### How to reproduce this bug?

1. Go to settings
2. Go to webcams
3. Click add webcam
4. Fill out all details and the name
5. Click save webcam

### Additional information:

I'm using a Qidi Q1 Pro, Mainsail is installed on my Raspberry Pi 4

meteyou, 2024-09-08:
which moonraker version do you use?

Rodin10, 2024-09-08:
I use Moonraker v0.9.3-0-g71f9e677

meteyou, 2024-09-08:
pls post also a screenshot from your webcam settings which you try to store

Rodin10, 2024-09-08:
[image]

meteyou, 2024-09-08:
i tested exactly the same without any issue. i only know this error with an outdated moonraker version, but your moonraker version looks up-to-date.

meteyou, 2024-09-09:
is it possible, that you use an extra Pi with Mainsail & Moonraker and then connect to the Qidi pro? so you use the default Klipper & Moonraker on the Qidi pro?

Rodin10, 2024-09-09:
Not quite sure what you mean unfortunately

Rodin10, 2024-09-09:
I just noticed that when connecting to my Q1 Pro the Moonraker version changes...
How do I update that?

[image]

meteyou, 2024-09-09:
yes, i exactly mean this. if you join to another printer, you connect to Moonraker from this printer. so you only have Mainsail from your first connection.

the Moonraker version from Qidi is too old (you should also see a warning in the notification center [the bell in the topbar]). So please update Moonraker on your printer to fix this issue.

closed 2024-09-09 by meteyou