mainsail-crew/mainsail #2008: Trigger spoolman prompts from load/unload filament macros

bo0tzz, 2024-09-12:
### Requested feature:

Add the option to automatically eject the currently selected spool from an UNLOAD_FILAMENT macro, and to trigger a prompt to select a spool from the LOAD_FILAMENT macro.

### Solves the following problem:

I sometimes forget to change the selected spool when changing it out.

### Additional information:

An alternative approach would be to integrate filament load/unload more deeply with the spoolman feature. That could also allow for smartly (un)loading at the appropriate temperature for the selected spool.

If someone has some pointers for what code I would need to look at, I'd be happy to try my hand at a PR for this.

meteyou, 2024-09-12:
hey! thx for you request.

you can just use the macro prompts to implement that with your config: https://docs.mainsail.xyz/overview/features/macro-prompts

maybe this repo could also help you: https://github.com/spooknik/FLAP

bo0tzz, 2024-09-12:
The macro prompts look really cool and pretty close, but I'm not sure how I would get those to show the spool pick modal?

FLAP looks nice, thanks for linking that!

meteyou, 2024-09-12:
> The macro prompts look really cool and pretty close, but I'm not sure how I would get those to show the spool pick modal?

This is not possible with it...

bo0tzz, 2024-09-13:
I guess the ability to show that is one of the options I'm asking for then. 

If triggering the spoolman prompt from a macro isn't an option, but ejecting the spool is, then another approach could be a setting to refuse starting a print if a spool isn't selected.

meteyou, 2024-09-13:
ejecting the spool is just a "gcode". but opening the dialog can create other issues. for example: when we add an option to open this dialog, how do you want to create a "clean solution" to close it? for example you have to write an output on your console (like we did with the macro prompts), but then every open gui (or guis you will open after that) will automaticaly open this dialog. it doesnt matter if you have chosen one spool in any open interface. so how do you want to fix this issue?

bo0tzz, 2024-09-13:
Yeah I guess that makes sense. How do you feel about the alternative approach of extending the spoolman panel on the dashboard to do filament load/unload in tandem with selecting/ejecting a spool?

meteyou, 2024-09-13:
ohh... I haven't thought about this idea until now, but I think this is the much better solution to fix this issue!

bo0tzz, 2024-09-13:
I imagine that as having spool eject or select call a configurable macro, which the user could then set up to do a filament load/unload. The extruder temperature from spoolman could get passed to the macro as a parameter. What do you think?