mainsail-crew/mainsail #2013: Gcode files on the main dashboard?

Stooovie, 2024-09-14:
### Requested feature:

Can we get Gcode file browser on the main dashboard, like Fluids has? And/or an option to have the sidebar always uncollected (visible)? Right now there is an option “navigation default state” that doesn’t seem to do anything - the panel always collapses even with “always open” selected. Because of this, picking a file to print takes three clicks instead of one. 

Thanks.

meteyou, 2024-09-14:
pls dont mix up multiple questions in one FR.

> Can we get Gcode file browser on the main dashboard, like Fluids has?

we wrote that already multiple times. we will not add a full gcode panel on the dashboard, because it is the most time complete useless (after you start a print).

> And/or an option to have the sidebar always uncollected (visible)? Right now there is an option “navigation default state” that doesn’t seem to do anything - the panel always collapses even with “always open” selected.

"navigation default state" is the state when you open Mainsail. your navigation is only "closed/collapses" when you use a small display and it is only in "overlay mode".

closed 2024-09-14 by meteyou

Stooovie, 2024-09-14:
Well, Fans and pins and Gcode viewer are completely useless for me and that option is there... Why not give the option to the user?

meteyou, 2024-09-14:
Fans and pins have only this panel and there is no other more detail special page for it.

> Why not give the option to the user?

because i see no possible way to implement it very useful and it is my freetime, which i spend in this project. nobody pays me for this job (expect some sponsors)...

i also implemented the latest files and printqueue already on the dashboard, which i think it is a way better solution instead of adding a full file explorer on the dashboard.