mainsail-crew/mainsail #2022: Color Picker not displaying for PCA9632 with no pins specified

dockterj, 2024-10-02:
### Mainsail Version:

2.12.0

### Browser:

Edge

### Device:

Laptop

### Operating System:

Windows

### What happened?

I have multi color LEDs connected to a PCA9632.  This device is connected by I2C using all default values:
`[pca9632 led]
color_order: BGRW
initial_BLUE: 1.0
initial_GREEN: 1.0
initial_RED: 1.0`

This led shows up in Mainsail under miscellaneous, but I am only able to pick the blue channel.

### What did you expect to happen instead?

Some kind of multi color picker should appear.

### How to reproduce this bug?

From discord conversations it appears that not having pins specified is causing this device to not be recognized properly.

meteyou, 2024-10-15:
hey @dockterj, pls upload a klippy.log

meteyou, 2024-10-15:
you can also test the PR #2028 if you have time. I think  the issue was, because PCA9632 doesnt return a array for the color order.

dockterj, 2024-10-15:
@meteyou thanks for this.  I'll be home in a few days and will test as soon as I can.

meteyou, 2024-10-15:
@dockterj thank you very much! if you need "faster help", feel free to ping me on discord.

dockterj, 2024-10-17:
http://klipperbot.local/printer/objects/query?pca9632%20led

{
    "result": {
        "eventtime": 2353.670446115,
        "status": {
            "pca9632 led": {
                "color_data": [
                    [
                        1,
                        1,
                        1,
                        0,
                        0
                    ]
                ]
            }
        }
    }
}

dockterj, 2024-10-17:
[klippy (58).log](https://github.com/user-attachments/files/17413061/klippy.58.log)

dockterj, 2024-10-17:
I tried changing the color order in printer.cfg and it appears that mainsail is picking the first color.  Is this a bug in the PCA9632 implementation that it isn't returning the color order array?

meteyou, 2024-10-17:
pls post also the content from `http://klipperbot.local/printer/objects/query?configfile=settings`.

the status return is a little bit different then a neopixel return. neopixels return an array with the colorOrder, but the PCA9632 only return a string. you can also test the PR #2028 . there i already tried to fixed this issue.

can you pls explain which hardware do you use? maybe i can buy something like that to test this. maybe there is also an issue with the LED templates, but i cannot test it without the hardware.

dockterj, 2024-10-19:
[query.json](https://github.com/user-attachments/files/17447054/query.json)

dockterj, 2024-10-19:
I maintain a fork of Klipper to run on Makerbot Replicator 2 and 2X printers (and other clones that use type k thermocouples and ADS1118 for the hotend).  

I'll test the PR you linked to - I don't have an environment set up yet for Mainsail development - I'm working on that right now.

meteyou, 2024-10-19:
@dockterj you can download a "prebuild" zip from the workflow. you don't have to set up a mainsail development

dockterj, 2024-10-19:
@meteyou Sorry I'm not familiar enough with Github to know where to get that from.  Can you send me a link?

meteyou, 2024-10-19:
on this page in the "Artifacts" panel is a "mainsail-latest.zip": https://github.com/mainsail-crew/mainsail/actions/runs/11352773937?pr=2028

this is the build from the PR workflow. Just unzip it and upload it via SFTP to ~/mainsail (pls remove all files in this directory, before you upload the new files).

a force reload (SHIFT+CTRL+F5) will be also needed, to load the new files in your browser.

PR #2028 (2024-11-07): fix: fix color picker for PCA9632
## Description

This PR fix the color picker for PCA9632. 

## Related Tickets & Documents

fixes #2022 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none

## Summary by Sourcery

Bug Fixes:
- Fix the color picker functionality for PCA9632 by correcting the handling of the color_order setting.
closed 2024-11-07 by meteyou