mainsail-crew/mainsail #2039: Twin y axis for separate range temperature

cvanelteren, 2024-11-08:
### Requested feature:

Dear all,

I am running into the issue sometime that my mainboard reaches unwanted temperatures due to the enclosure that the printer is in. I wish to monitor this more closely. However, it is rather hard to read the difference in temperature due to the scale difference between the mainboard temps and the extruder temp. Is it possible to add a twin y axis and plot certain temperatures on a separate range?

meteyou, 2024-11-08:
No, this is not possible. The second y axis for the chart is for the power (pwm) plot.

closed 2024-11-08 by meteyou

cvanelteren, 2024-11-08:
Can'

> No, this is not possible. The second y axis for the chart is for the power (pwm) plot.

Can't this be togleable? Or alternatively allow for log scales, or allow for breaks in the present ylabel?