mainsail-crew/mainsail #2044: Feat: Device Tab from within Settings

krautech, 2024-11-15:
### Requested feature:

[image]
[image]

Would it be neat to have this available in settings as well not just hidden away in the file browser behind a bunch of pages.
Simply have it as a tab within settings that lists all the devices in the window. I included examples.

### Solves the following problem:

The amount of people who dont know you can view all your connected devices from within mainsail is astounding. Also having it accessible through settings gives it more opportunity to be utilized.

flyespresso, 2024-11-15:
Oh heck yeah this would be a welcome QOL improvement, it's been maddening to use so far in it's current incarnation.

adelyser, 2024-11-15:
Would love to see a devices tab to easily access this feature. I think it would be worth mentioning when lauched from the devices tab that it's also accessible from the config file editor. Kinda like a "Did you know....." type of thing.

rackrick, 2024-11-15:
Hey guys...
may I ask you what are you using this feature for?

Becaus our intension was, to make the device paths easily accessible, while writing your config. When its in die settings panel this benefit is gone. So what is your goal when you access it through the settings panel?

krautech, 2024-11-15:
For me, I find I use it alot both in when I'm writing Configs, and when I'm diagnosing issues for myself or with others.

It's much easier to go into settings and then see an entire device list than try explain how to find it to people who aren't overly experienced with mainsail.

Having it in both just makes it a bit easier to "stumble" across and know it's there. 

I spoke with a few people before writing this feature up and I think maybe 50% didn't know there was even a page that did this but could tell me what was in the settings pages 😂

I don't think it's "ground-breaking“ but it might be something that's conveniently more there, for people to find and use.

Lindloven, 2024-11-15:
Yeah this would be an awesome feature. Makes it alot easier!

meteyou, 2024-11-15:
@Lindloven can you pls describe this:
> Makes it alot easier!

meteyou, 2024-11-15:
@krautech this:
> In the end I ended up making a video to send to them showing which buttons to click to find the devices list and they were shocked they'd never seen it before and that it's likely because it's buried quite deep into the UI.

will not be fixed by this change... we had something like that also before and users said "this will fix it" and "this will fix it", but you will ever find a user, which will say "i never found it!".

to be honest. another position for the identical function is in my eyes just "useless". and the interface section needs another structure, so we cannot reuse the device dialog and we would have to "reimplement" it a second time. this would be the worst we could do...

so if we would refactor it and just move (so not on both places at the same time), do you think it would be the better place for every use case inside of the interface settings?

meteyou, 2024-11-15:
> Oh heck yeah this would be a welcome QOL improvement, it's been maddening to use so far in it's current incarnation.

can you please describe this? how do you use device paths in your editor? do you think it is easier to exit the editor, open the interface settings, open the device tab, search for the device, close all interface settings, open the editor again is an easier use case?

krautech, 2024-11-15:
Yeah you're not wrong, there's always something 😂 

Ahhh interesting that you can't reuse it hmm that does complicate things a little.

I haven't looked at the mainsail source code so wasn't sure if it was as simple as recalling it or not.

I don't think removing it from the code editor makes sense sadly. Since it is good to also have there. (atleast in my opinion)

Is there any other possible solutions to easily do it? Or perhaps a better place to have it solely? 🤔

krautech, 2024-11-15:
> can you please describe this? how do you use device paths in your editor? do you think it is easier to exit the editor, open the interface settings, open the device tab, search for the device, close [...]

I don't think in this situation that's what's happening.

I think it's more being used for debugging purposes, helping flash devices etc.

That's why my first thought would be to have it in both locations since ideally, it has two main use cases.

rackrick, 2024-11-15:
> In the end I ended up making a video to send to them showing which buttons to click to find the devices list and they were shocked they'd never seen it before and that it's likely because it's buried quite deep into the UI.

Well if you had to this far and create a user to follow your clicks... I am not sure if a different position of same function would help here in anyways. Thats an "layer 8" issue... if you know what I mean.

> because it's buried quite deep into the UI.
if you "bury" it inside the editor or the user settings is quite the same story :

`Open Settings -> Click on devices section` vs. `Open Machiene Page -> Open File -> Click devices`

It just a click more ore less. I don't belive it will make a big difference.
In the all what you guys describe leads to a change in the config files. Which means you need to open the editor again.

We could refactor it to fit both places but right now I can't see the improvement here.

@meteyou to make the access more easier we might could add a direct button on the machiene page?
That would at least fit the "topic" of the page  and could be used to access it faster for troubleshooting.

meteyou, 2024-11-15:
> Ahhh interesting that you can't reuse it hmm that does complicate things a little.
I haven't looked at the mainsail source code so wasn't sure if it was as simple as recalling it or not.

it would be no issue to open the same small dialog on another place. for example we add a button in the system panel to open it. but the interface settings dialog has a different structure, where this dialog would not fit.

> Is there any other possible solutions to easily do it? Or perhaps a better place to have it solely?

it would work on every place, where we only have a button to open the dialog. then it would be reuse able.

> Some of the others might think it's better in the interface screen though.

to be honest. the interface settings only contains mainsail (interface) settings. so this kind of options would be complete wrong in there. devices have nothing todo with the interface...

> I think it's more being used for debugging purposes, helping flash devices etc.
That's why my first thought would be to have it in both locations since ideally, it has two main use cases.

Having something in multiple positions looks also wrong for me... and if you need it after flashing, the next step is to setup the config. so if you use it in the editor, you could also do the next step...

meteyou, 2024-11-15:
@rackrick 

> We could refactor it to fit both places but right now I can't see the improvement here.

that will not work with the interface settings, because of the tab layout...

meteyou, 2024-11-15:
the machine page and maybe the "system panel" would be in my mind a much better place for this "context".

krautech, 2024-11-15:
Looking over the machine page, I agree.

PR #2046 (2024-11-23): feat: add button to open the device dialog in SystemPanel
## Description

This PR adds a button to open the DeviceDialog in the SystemPanel in the Machine page.

## Related Tickets & Documents

fixes #2044 

## Mobile & Desktop Screenshots/Recordings

[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-11-23 by meteyou