mainsail-crew/mainsail #2052: Spoolman integration search function Broken on Brave browser

Ntrondle, 2024-11-23:
### Mainsail Version:

v2.12.0

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

When trying to use the search function in mainsail the interface locks up and will not let the user browse through the different pages of spool, the search function will also not fetch the filament that was entered in the search window. 

The issue is fixed by reloading the entier page.

### What did you expect to happen instead?

search function should work and not freeze the spoolman spool selection page.

### How to reproduce this bug?

Reported not working on Brave :
Version **1.71.121** Chromium: **130.0.6723.91** (arm64)

sppolman : 
Version 0.20.0 (4f8d5ab)

### Additional information:

let me know if additional information should be given. happy to give anything else.

Ntrondle, 2024-11-23:
Additional testing with Safari, Firefox, and Chrome shows the same behavior as observed with Brave.

meteyou, 2024-11-24:
@Ntrondle thx for your report. i updated spoolman right no to v0.20.0 and cannot find any issues. pls upload your moonraker.log. maybe i can find any issue in it.

Ntrondle, 2024-11-27:
sorry for the late reply...

here's my moonraker log hopefully you can find the reason behind my problem as it is quite odd...

[moonraker.log](https://github.com/user-attachments/files/17937539/moonraker.log)

meteyou, 2024-11-30:
i checked your moonraker.log and found this:
```
2024-11-27 13:18:27,844 [spoolman.py:_connect_websocket()] - Connecting To Spoolman: ws://192.168.2.24:7912/api/v1/spool
2024-11-27 13:18:27,850 [server.py:add_log_rollover_item()] - Failed to Connect to spoolman: [Errno 101] Network is unreachable
```
I also found this issue in your log:
```
2024-11-27 13:18:31,875 [announcements.py:_fetch_moonlight()] - Failed to update subscription 'klipper': [Errno -3] Temporary failure in name resolution
2024-11-27 13:18:33,889 [announcements.py:_fetch_moonlight()] - Failed to update subscription 'spoolman': [Errno -3] Temporary failure in name resolution
```
This have nothing todo with the spoolman issue itself, but it looks like a generic "network setting issue" here.

But i should check if i can catch "network issues" when using the spoolman API via moonraker.

closed 2024-12-23 by github-actions[bot]