mainsail-crew/mainsail #2062: *Feature Request - command to check job queue

Wetzel402, 2024-12-01:
### Requested feature:

Add a command that returns job queue length

### Solves the following problem:

Allows for macros to take the job queue into account. 

Example: END_PRINT could check the job queue length and if it returns 0 continue shutdown but if it returns >0 it pauses so the bed can be cleared and the user can resume.

meteyou, 2024-12-01:
Mainsail is only a Webinterface. if you close your browser, Mainsail is not running. So this feature cannot be added to Mainsail.

Every Command or Gcode is implemented in Klipper. So you have to add a macro in your Klipper config. But there is one problem. The jobqueue itself is a Moonraker feature. So I dont think you have access to the Moonraker jobqueue length. You can only start a shutdown with a delay and cancel it with the next print start. so it will only be executed, when no print is started. If you need help to set it up, feel free to use our discord. This is not the place for help itself.

Wetzel402, 2024-12-02:
For the record here is my test example gcode...

```
[delayed_gcode POWER_DOWN]
gcode:
  POWER_OFF_PRINTER

[gcode_macro CONTINUE]
gcode:
  UPDATE_DELAYED_GCODE ID=POWER_DOWN DURATION=0.1
  RESPOND type="command" msg="action:prompt_end"
  
[gcode_macro CANCEL]
gcode:
  UPDATE_DELAYED_GCODE ID=POWER_DOWN DURATION=0
  RESPOND type="command" msg="action:prompt_end"

[gcode_macro SHOW_PROMT]
gcode:
  UPDATE_DELAYED_GCODE ID=POWER_DOWN DURATION=60
  RESPOND TYPE=command MSG="action:prompt_begin Shutdown"
  RESPOND TYPE=command MSG="action:prompt_text Printer shutting down. Cancel?"
  RESPOND TYPE=command MSG="action:prompt_footer_button continue|CONTINUE|primary"
  RESPOND TYPE=command MSG="action:prompt_footer_button CANCEL|CANCEL|error"
  RESPOND TYPE=command MSG="action:prompt_show"
```