mainsail-crew/mainsail #2068: Camera-Streamer WebRTC not working

Zeanon, 2024-12-06:
### Mainsail Version:

v2.13.0

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I updated mainsail to the current version (v2.13.0) and my webcam didnt work anymore, instead I got a spammed console:
[image]
tested on Vivaldi and Chrome

### What did you expect to happen instead?

Webcam showing my printer

### How to reproduce this bug?

use camera streamer webrtc

closed 2024-12-06 by github-actions[bot]

Zeanon, 2024-12-06:
Why was it closed immediately?
Using an older version of the webrtc class makes the cam work perfectly fine, so the cam itself works (in fluidd it also works)

meteyou, 2024-12-06:
@Zeanon i tested it and cannot reproduce any issue, so it looks like a setup issue. pls follow the description above.

why should i care here if it works in fluidd or not? this is the mainsail repo...

Zeanon, 2024-12-06:
> @Zeanon i tested it and cannot reproduce any issue, so it looks like a setup issue. pls follow the description above.
> 
> why should i care here if it works in fluidd or not? this is the mainsail repo...

I know it's the mainsail repo, I just mentioned fluidd to clarify it's not a camera issue(I did it to verify it's a mainsail issue since I don't want to report a bug that is not mainsail related on mainsail obviously)

meteyou, 2024-12-06:
@Zeanon its an outdated camera-streamer version... you would know it, when you would follow the description above...

Zeanon, 2024-12-06:
> @Zeanon its an outdated camera-streamer version... you would know it, when you would follow the description above...

My crowsnest is up to date 
How should I know the camera streamer is outdated?
I tend to not touch the backends so I don't break anything

meteyou, 2024-12-06:
crowsnest != camera-streamer...

i will block this issue  right now, because its no mainsail issue and help for something is on our discord server. thx for the report, but its not any topic for here...

PR #2069 (2024-12-07): fix(Webcam): add ICE check to support older CS versions
## Description

This PR adds a check for ICE Candidates to fix the connection to older camera-streamer backends (for example the default version for the latest MainsailOS v1.3.2).

## Related Tickets & Documents

fixes #2068 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none