mainsail-crew/mainsail #2073: Unable to open file when printing from home screen

Zchulski, 2024-12-08:
### Mainsail Version:

2.13.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I got an error when trying to print a file from the home screen.  "unable to open file"  after checking on discord somebody suggested trying to print the file from the G-code screen.  I was successfully able to print a file from the G-code screen.

### What did you expect to happen instead?

I expected to be able to start printing a file from the home screen instead of having to switch screens to print a file 

### How to reproduce this bug?

Go to home screen and try to print a file from there.  you should get an error that says unable to open file.
then go to the g-code screen and try to print from there

meteyou, 2024-12-08:
i tested it right now without any issue. can you pls give us a little bit more context?

Zchulski, 2024-12-08:
[klippy (14).log](https://github.com/user-attachments/files/18053143/klippy.14.log)

I have attached the klippy log.  infurunately I cant give any more info other than from the home screen when I try to print something I get an error that says unable to open file.  but if I try to print from the g-code screen it prints just fine

meteyou, 2024-12-08:
can you post the filename from the gcode file?

Zchulski, 2024-12-08:
table tennis.gcode

meteyou, 2024-12-08:
all files are in a subdirectory?

Morfesto, 2024-12-08:
I've had the same issue since upgrading to v2.13.0 and it persists in v2.13.1

I suspect it's something with the coding of the standby window in the dashboard.

Here are some examples from my moonraker log.

```
2024-12-06 19:46:59,940 [klippy_apis.py:start_print()] - Requesting Job Start, filename = I3 Parts/I3 Parts/V350_Motor Mount - Rod Mount v0.gcode
2024-12-06 19:46:59,959 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.print.start, Code: 400, Message: Unable to open file
```

Those two lines are from using the standby window in the dashboard
These are a successful start from the G-Code File menu

```
2024-12-06 19:51:39,124 [klippy_apis.py:start_print()] - Requesting Job Start, filename = I3 Parts/V350_Motor Mount - Rod Mount v0.gcode
2024-12-06 19:51:39,214 [job_state.py:_status_update()] - Job Started: I3 Parts/V350_Motor Mount - Rod Mount v0.gcode
```

Looks like the standby window in the dashboard is adding an extra folder

```
[8:12 PM]
I3 Parts/I3 Parts/V350_Motor Mount - Rod Mount v0.gcode <--vs--> I3 Parts/V350_Motor Mount - Rod Mount v0.gcode
```

Morfesto, 2024-12-08:
Looking at OPs moonraker log I see the same issue.

```
2024-12-08 11:03:47,426 [klippy_apis.py:start_print()] - Requesting Job Start, filename = 737 Overhead/737 Overhead/table tennis.gcode
2024-12-08 11:03:47,432 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.print.start, Code: 400, Message: Unable to open file
```

vs

```
2024-12-08 11:08:02,212 [klippy_apis.py:start_print()] - Requesting Job Start, filename = 737 Overhead/table tennis.gcode
2024-12-08 11:08:02,336 [job_state.py:_status_update()] - Job Started: 737 Overhead/table tennis.gcode

```

It's doubling the first sub directory  "737 Overhead/737 Overhead/"

I suspect that whatever code is in the standby window it's probably got a variable in twice accidentally.

Lexanger, 2024-12-08:
I'm glad I came across this! I thought it was something with my switch to a new pi and I've been feeling like a crazy person.

meteyou, 2024-12-08:
@Lexanger all files from the root directory will also work, or just print it from the gcode file page. there is also no issue with subdirectories.

PR #2074 (2024-12-09): fix: fix print start from dashboard for subdirectory files
## Description

This PR will fix the issue with start prints form the dashboard with files in a subdirectory. It creates a new file object without the complete path as filename for the start-print-dialog.

## Related Tickets & Documents

fixes #2073 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-12-09 by meteyou