mainsail-crew/mainsail #2098: Arrow next to Z Thermal Adjust, indicating ascending vs descending temp.

PhilBaz, 2024-12-27:
### Requested feature:

Simple idea. An arrow next to the Z Thermal Adjust (in the temp panel) that would indicate if the temp is moving up or down.

The problem is that when the printer is (re)heating, after a previous print, it can be difficult to know whether the Z Thermal Adjust is actually moving down or up, since it takes a long time for this to turn around. It would be great to have an arrow to show the direction of travel. 

[image]

### Solves the following problem:

Don't know if the frame temp is actually ascending or descending.

### Additional information:

Hope this makes sense... The basic theory is that bed meshing is more reliable when frame temp is moving up, rather than changing directions after the bed mesh is taken. 

Let me know if more info would be helpful.

meteyou, 2024-12-27:
hey @PhilBaz! what should the algorithm for the symbol look like? because Moonraker sends the data at 200ms intervals, it would mostly display "the same" and then only briefly up or down. therefore, a more "advanced algorithm" would make more sense here.

PhilBaz, 2024-12-30:
If possible maybe average of 30 or 40 seconds of Moonraker temp, then compare that against the next 30-40 seconds. If posible maybe a rolling average of  n-40 to n-80 seconds, subtracted from n-39 to n-0. if that makes sense?

I think part of what you are saying is that the symbol would only show for some amount of time when the temp had not stabilized. Like it would be there when rising or falling but not when relatively stable? Would it possibly make sense to just always have the last direction indicated even if it has stabilized?

Thanks for considering the request :)

meteyou, 2024-12-30:
> I think part of what you are saying is that the symbol would only show for some amount of time when the temp had not stabilized. Like it would be there when rising or falling but not when relatively [...]

@PhilBaz no, i didn't mean to say that. i just wanted to ask what the relevant range for calculating the direction should be. if i only query to the last data point, it would probably also show 90% of the time == and not up or down, because the temperature change is much too slow.

PhilBaz, 2025-01-14:
@meteyou I see this is going stale... Didn't know if you would like me to leave it be. Did the approach above make any sense in terms of getting a useful symbol?