mainsail-crew/mainsail #2100: After updating to 2.13.2 no tool head in the dashboard Creality K1Max and K1

Atrox101, 2024-12-27:
### Mainsail Version:

v2.13.2

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

After I did the latest update on my Root Klipper K1 Max, I noticed that the tool head is no longer displayed. 

The error was not present on another printer on which I also carried out this update, which is completely custom. It must be a pure problem in connection with the root with Creality printers.

A friend with a K1 had exactly the same problem after the update.

### What did you expect to happen instead?

That it works as usual. It worked before that too. Incidentally, I also looked in the settings at the cogwheel at the top right. The tap is also empty there.

### How to reproduce this bug?

Update

### Additional information:

[image]
[image]
[klippy.log](https://github.com/user-attachments/files/18263994/klippy.log)

meteyou, 2024-12-27:
pls post a screenshot from your browser console

Atrox101, 2024-12-27:
[image]
Like this?
I'm doing this for the first time. Incidentally, it's the same with Orca when I go under the device.

meteyou, 2024-12-27:
pls click on the console tab and post a screenshot in this area

Atrox101, 2024-12-27:
[image]

meteyou, 2024-12-27:
Thank you! I think I know what the issue is. Creality use a very old Klipper version.

Atrox101, 2024-12-27:
What I notice and my friend has exactly the same problem is you have to restart the printer completely but stay in the window then it comes back. But as soon as you close the window and reconnect to the printer it is gone.

tkunchick, 2024-12-28:
What ever it is, it affects both the toolhead controls, as well as multi extruders. if you have multi extruders setup, they do not appear in the extruder section after update. I reverted back to version 2.13.0  to correct the issue.

Running a K1 Max with Happy Hare and 4 colors.

Atrox101, 2024-12-28:
> Was auch immer es ist, es betrifft sowohl die Werkzeugkopfsteuerung als auch Multi-Extruder. Wenn Sie Multi-Extruder eingerichtet haben, werden diese nach dem Update nicht im Extruderbereich angezeigt. Ich bin auf Version 2.13.0 zurückgegangen, um das Problem zu beheben.
> 
> Betreibe einen K1 Max mit Happy Hare und 4 Farben.

How did you reset the version? Exact explanation.

tkunchick, 2024-12-28:
> How did you reset the version? Exact explanation.

I edited the paths.sh in the k1 helper script and directed the mainsail URL to https://github.com/mainsail-crew/mainsail/releases/download/v2.13.0/mainsail.zip

[paths.sh.zip](https://github.com/user-attachments/files/18266360/paths.sh.zip)
just unzip it and place it in your usr/data/helper-script/scripts overwriting the existing. 

use the helper script and uninstall mainsail, then reinstall it.

meteyou, 2024-12-28:
pls post the output of `http://<ip or hostname>/printer/objects/query?gcode`

tkunchick, 2024-12-28:
> pls post the output of `http://<ip or hostname>/printer/objects/query?gcode`

`{"result": {"eventtime": 68289.223161833, "status": {"gcode": {}}}}`

on mainsail v2.13.0

`{"result": {"eventtime": 68405.595008333, "status": {"gcode": {}}}}
`
on mainsail v2.13.3

meteyou, 2024-12-28:
thx! can you also post the output of `http://<ip or hostname>/printer/objects/list`

tkunchick, 2024-12-28:
> thx! can you also post the output of `http://<ip or hostname>/printer/objects/list`
> 
> PS: these links are only Moonraker API URLs. these have nothing todo with mainsail

404 not found for both versions

meteyou, 2024-12-28:
ok. doublechecked the url and works fine. pls check if you make some typo or something. (as i wrote above, these URLs have nothing todo with mainsail! its only a moonraker api call!)

tkunchick, 2024-12-28:
> ok. doublechecked the url and works fine. pls check if you make some typo or something. (as i wrote above, these URLs have nothing todo with mainsail! its only a moonraker api call!)

[image]

meteyou, 2024-12-28:
pls check if your browser redirect to https and fix it with http. maybe this is your issue

meteyou, 2024-12-28:
is Moonraker reachable via IP without any port in your case? i dont know the current K1 setup. in a normal MainsailOS/KIAUH instance, all URLs with /printer, /server and some more are redirected to Moonraker (port 7125).

but if this not work, i'm dont know, why the url above works without a port...

tkunchick, 2024-12-28:
my bad, didn't remove greatly web, and didn't add port.

`{"result": {"objects": ["webhooks", "configfile", "mcu", "mcu mmu", "mcu nozzle_mcu", "mcu leveling_mcu", "mcu rpi", "virtual_pins", "gcode_macro xyz_ready", "gcode_macro _IF_HOME_Z", "gcode_macro _IF_MOVE_XY", "gcode_macro _HOME_X", "gcode_macro _HOME_Y", "gcode_macro _HOME_Z", "gcode_macro PRINTER_PARAM", "gcode_macro AUTOTUNE_SHAPERS", "gcode_macro LOAD_MATERIAL_CLOSE_FAN2", "gcode_macro LOAD_MATERIAL_RESTORE_FAN2", "gcode_macro SET_E_MIN_CURRENT", "gcode_macro RESTORE_E_CURRENT", "gcode_macro LOAD_MATERIAL", "gcode_macro QUIT_MATERIAL", "gcode_macro Qmode", "gcode_macro Qmode_exit", "gcode_macro M204", "gcode_macro M205", "gcode_macro M900", "gcode_macro PRINT_CALIBRATION", "gcode_macro FIRST_FLOOR_PAUSE_POSITION", "gcode_macro ACCURATE_G28", "gcode_macro PRINT_PREPARED", "gcode_macro PRINT_PREPARE_CLEAR", "gcode_macro END_PRINT_POINT_WITHOUT_LIFTING", "gcode_macro END_PRINT_POINT", "gcode_macro END_PRINT", "gcode_macro FIRST_FLOOR_PAUSE", "gcode_macro FIRST_FLOOR_RESUME", "gcode_macro PAUSE", "gcode_macro INPUTSHAPER", "gcode_macro BEDPID", "gcode_macro TUNOFFINPUTSHAPER", "gcode_macro RESUME", "gcode_macro CANCEL_PRINT", "gcode_macro G29", "gcode_move", "print_stats", "fan_feedback", "custom_macro", "gcode_macro product_param", "gcode_macro _KLIPPER_BACKUP", "gcode_macro _KLIPPER_RESTORE", "gcode_macro _MOONRAKER_BACKUP", "gcode_macro _MOONRAKER_RESTORE", "gcode_macro KLIPPER_BACKUP_CONFIG", "gcode_macro KLIPPER_RESTORE_CONFIG", "gcode_macro MOONRAKER_BACKUP_DATABASE", "gcode_macro MOONRAKER_RESTORE_DATABASE", "gcode_macro RELOAD_CAMERA", "gcode_macro BED_LEVELING", "gcode_macro PID_BED", "gcode_macro PID_HOTEND", "gcode_macro WARMUP", "calibrate_shaper_config", "gcode_macro INPUT_SHAPER_CALIBRATION", "gcode_macro TEST_RESONANCES_GRAPHS", "gcode_macro BELTS_SHAPER_CALIBRATION", "gcode_macro EXCITATE_AXIS_AT_FREQ", "heaters", "output_pin ADAPTIVE_BED_MESH", "output_pin FULL_BED_MESH", "output_pin ADAPTIVE_PURGE_LINE", "gcode_macro _ADAPTIVE_BED_MESH_ON", "gcode_macro _FULL_BED_MESH_ON", "gcode_macro _BED_MESH_OFF", "gcode_macro _ADAPTIVE_PURGE_LINE_ON", "gcode_macro _ADAPTIVE_PURGE_LINE_OFF", "gcode_macro KAMP_BED_MESH_SETTINGS", "gcode_macro KAMP_PURGE_LINE_SETTINGS", "gcode_macro START_PRINT", "gcode_macro BED_MESH_CALIBRATE", "gcode_macro _LINE_PURGE", "gcode_macro _SMART_PARK", "gcode_macro DEFINE_OBJECT", "gcode_macro START_CURRENT_OBJECT", "gcode_macro END_CURRENT_OBJECT", "gcode_macro LIST_OBJECTS", "gcode_macro LIST_EXCLUDED_OBJECTS", "gcode_macro REMOVE_ALL_EXCLUDED", "gcode_macro _KAMP_Settings", "gcode_macro GET_TIMELAPSE_SETUP", "gcode_macro _SET_TIMELAPSE_SETUP", "gcode_macro TIMELAPSE_TAKE_FRAME", "gcode_macro _TIMELAPSE_NEW_FRAME", "gcode_macro HYPERLAPSE", "gcode_macro TIMELAPSE_RENDER", "gcode_macro TEST_STREAM_DELAY", "gcode_macro _MMU_CUT_TIP", "gcode_macro _FILAMETRIX_MOVE_IN_BOUNDS", "gcode_macro _FILAMETRIX_MOVE_TO_CUTTER_PIN", "gcode_macro _FILAMETRIX_DO_CUT_MOTION", "gcode_macro _FILAMETRIX_GANTRY_SERVO_DOWN", "gcode_macro _FILAMETRIX_GANTRY_SERVO_UP", "gcode_macro _MMU_FORM_TIP", "gcode_macro _WAIT_FOR_TEMP", "tmc2209 stepper_mmu_gear", "tmc2209 stepper_mmu_gear_1", "tmc2209 stepper_mmu_gear_2", "tmc2209 stepper_mmu_gear_3", "filament_switch_sensor mmu_pre_gate_0", "filament_switch_sensor mmu_pre_gate_1", "filament_switch_sensor mmu_pre_gate_2", "filament_switch_sensor mmu_pre_gate_3", "filament_switch_sensor mmu_gate_sensor", "filament_switch_sensor extruder_sensor", "mmu_sensors", "gcode_macro _MMU_LED_ACTION_CHANGED", "gcode_macro _MMU_LED_PRINT_STATE_CHANGED", "gcode_macro _MMU_LED_GATE_MAP_CHANGED", "gcode_macro _MMU_SET_LED", "gcode_macro _STOP_LED_EFFECTS", "gcode_macro _SET_LED_EFFECT", "save_variables", "gcode_macro _MMU_SOFTWARE_VARS", "gcode_macro _MMU_STATE_VARS", "gcode_macro _MMU_LED_VARS", "gcode_macro _MMU_SEQUENCE_VARS", "gcode_macro _MMU_CUT_TIP_VARS", "gcode_macro _MMU_FORM_TIP_VARS", "gcode_macro _MMU_CLIENT_VARS", "gcode_macro T0", "gcode_macro T1", "gcode_macro T2", "gcode_macro T3", "query_endstops", "motion_report", "tmc2209 extruder", "mmu", "gcode_macro _MMU_PARK", "gcode_macro _MMU_SAVE_POSITION", "gcode_macro __MMU_SAVE_POSITION", "gcode_macro _MMU_RESTORE_POSITION", "gcode_macro _MMU_RETRACT", "gcode_macro _MMU_UNRETRACT", "gcode_macro _MMU_CLEAR_POSITION", "gcode_macro MMU_UPDATE_HEIGHT", "gcode_macro _MMU_AUTO_HOME", "gcode_macro _MMU_PRE_UNLOAD", "gcode_macro _MMU_POST_FORM_TIP", "gcode_macro _MMU_POST_UNLOAD", "gcode_macro _MMU_PRE_LOAD", "gcode_macro _MMU_POST_LOAD", "gcode_macro _MMU_ERROR", "gcode_macro _MMU_LOAD_SEQUENCE", "gcode_macro _MMU_UNLOAD_SEQUENCE", "gcode_macro MMU_START_SETUP", "gcode_macro MMU_START_CHECK", "gcode_macro MMU_START_LOAD_INITIAL_TOOL", "gcode_macro MMU_END", "gcode_macro MMU_COLD_PULL", "gcode_macro _MMU_ERROR_DIALOG", "gcode_macro _MMU_RUN_MARKERS", "gcode_macro MMU__UNLOAD", "gcode_macro MMU__EJECT", "gcode_macro MMU__HOME", "gcode_macro MMU__STATUS", "gcode_macro MMU__MOTORS_OFF", "gcode_macro MMU__SERVO", "gcode_macro MMU__SELECT_TOOL", "gcode_macro MMU__SELECT_BYPASS", "gcode_macro MMU__LOAD_BYPASS", "gcode_macro MMU__RECOVER", "gcode_macro MMU__PRELOAD", "gcode_macro MMU__CHECK_GATE", "gcode_macro MMU_CHANGE_TOOL_STANDALONE", "gcode_macro MMU_CHECK_GATES", "gcode_macro MMU_REMAP_TTG", "gcode_macro MMU_FORM_TIP", "gcode_macro _MMU_PRINT_START", "gcode_macro _MMU_PRINT_END", "gcode_macro _MMU_UPDATE_HEIGHT", "gcode_macro _MMU_ACTION_CHANGED", "g [... cut ...] n", "temperature_sensor _mcu_temp", "temperature_sensor _chamber_temp", "pause_resume", "filament_switch_sensor filament_sensor", "heater_fan hotend_fan_&_Bento_Cleaner", "idle_timeout", "virtual_sdcard", "tmc2209 stepper_x", "tmc2209 stepper_y", "tmc2209 stepper_z", "heater_bed", "output_pin fan0", "output_pin fan1", "output_pin fan2", "output_pin LED", "output_pin power", "probe", "bed_mesh", "display_status", "exclude_object", "system_stats", "manual_probe", "toolhead", "extruder"]}}`
v2.13.3

meteyou, 2024-12-28:
as i wrote multiple times above... these are moonraker api calls! so you dont have to post "both" outputs. these will everytime exactly the same!

tkunchick, 2024-12-28:
> as i wrote multiple times above... these are moonraker api calls! so you dont have to post "both" outputs. these will everytime exactly the same!

I am slow sometimes :-) PC tech, not a programmer, just wanted to make sure I was giving you all the info required.

meteyou, 2024-12-28:
these 2 outputs should help to fix this issue. i will post a short guide to "update" to the dev build. i would be happy, when you can verify it. its very difficult for me to test, because i dont have a printer with a crealityOS.

meteyou, 2024-12-28:
@tkunchick you can download the new test build here: https://github.com/mainsail-crew/mainsail/actions/runs/12528617227?pr=2102

your workflow wont work, because you have to be logged in GitHub + GitHub make a zip for the zip file... so you have to extract it 2 times (thx to GitHub for this...)

tkunchick, 2024-12-28:
so I can't just set the URL for https://github.com/meteyou/mainsail-dev/releases/download/v2.9.8/mainsail.zip ?

meteyou, 2024-12-28:
v2.9.8 is a complete other version and my link above is only a CI test build. no release.

tkunchick, 2024-12-28:
sadly, I don't understand. I am going to be out for 2-3 hours before I can get back to this.

will put my google skills to the test to figure out part of GitHub.

meteyou, 2024-12-28:
you can only download the "mainsail-latest.zip" from my link above, unzip it twice and upload it via ssh (filezilla or winscp or similar)

tkunchick, 2024-12-28:
Ok, so I updated to the CI, still reports as version 2.13.0 (did I install the right one?), but all UI elements are showing.

meteyou, 2024-12-29:
The CI build should show 2.13.2 (the current release number). If you don't see it your update didn't work.

meteyou, 2024-12-29:
I double-checked the PR and it use the v2.13.2:

[image]

tkunchick, 2024-12-29:
this file at the bottom correct?
[image]

tkunchick, 2024-12-29:
OK thing I got it. had to change the permissions on all the files after uploading..

[image]

Toolhead window seems fixed and stays after manually refreshing several times.
[image]

but the extruder window is still missing the 4 tools.
[image]

meteyou, 2024-12-29:
Thanks! Yes, I also updated the check to read for tools in this release. I'm also surprised that you're using such an old Klipper version with a rooted Creality printer. Is there no option to update the Klipper version?

tkunchick, 2024-12-29:
The only thing I have seen so far that seems viable is 

https://github.com/pellcorp/creality/wiki/Simple-AF

I don't have a spare bl touch or other bed leveling sensor and it requires a lot of hardware changes to the bed that I am not ready to delve into yet.

I don't know if I was clear or not, I have a 4 extruder system, with the update from yesterday, they do not show.

I am still running 2.13.0 to keep the tools available.

meteyou, 2024-12-29:
@tkunchick I pushed another commit to add another fallback for older klipper versions to show the toolchange macros.

you can find the testbuild here: https://github.com/mainsail-crew/mainsail/actions/runs/12539037131

tkunchick, 2024-12-30:
[image]
That seems to have gotten it as far as I can tell.

Thank you very much for your hard work.

Kaelum, 2025-01-02:
@meteyou just a heads up, but Creality made heavy changes to Klipper to get it to work, including safety features that do not exist in main line Klipper.  AFAIK, for whatever reason, Klipper hasn't accepted the changes that Creality has made.  If we move to main line Klipper, our printers will not work.  BEST

meteyou, 2025-01-02:
@Kaelum I never saw any PR from creality to Klipper. So I thing they won't merge any back to Klipper.

I just saw how they implemented Marcos and other functions in Klipper on a K1 and it was horrible...

mryel00, 2025-01-02:
> including safety features that do not exist in main line Klipper. AFAIK, for whatever reason, Klipper hasn't accepted the changes that Creality has made.

Can you please link the source of your information?

Kaelum, 2025-01-02:
This was over 18 months ago, when Creality was talking about releasing the code here.  I only know that there were discussions, and for whatever reason, nothing was merged.  I never said that a PR was issued.

meteyou, 2025-01-02:
@Kaelum creality wrote that they have created there own firmware and they don't use Klipper. Then some users hacked and published there code, that creality couldn't "hide" the fact any more. Then they just published there code themselves to fixed licence issues from/with Klipper. 

So there was never any attention from creality to merge anything back to mainline...

Kaelum, 2025-01-02:
@meteyou those are false statements.  Please provide citations.

meteyou, 2025-01-02:
@Kaelum loool really? It's written everywhere! just use google and you will find many references about this "shitstorm" when the K1 was published. there are a lot of news blogs with this infos.

you dont provide anything, but i have to citations? really? that's why we were so curious about your sources for your false statements!

Kaelum, 2025-01-02:
@meteyou it seems that all you want is an argument.  I will not provide you with that.  The very first public information from Creality in regard to the K1 line of printers, very specifically stated that they were based on Klipper.  Any statements contrary to that are false.  BEST

meteyou, 2025-01-02:
@Kaelum No, they're just making things up. Creality only called it "CrealityOS". There was NEVER any mention of Klipper here. it was only after the hack that Creality called it Klipper + published it. please check the timestamps of videos and Creality social media posts again. Why else would there be a huge shitstorm against Creality if they had done it "right"?

Furthermore, this discussion has no place here. I can also simply close the PR and block any implementation for "CrealityOS" in Mainsail. I don't need it and don't want to support it. I only do it because users ask me to.

Creality itself is the worst / most evil company in the 3d printer market, which only uses OpenSource and gives nothing back!

So pls provide data about your statement:
> AFAIK, for whatever reason, Klipper hasn't accepted the changes that Creality has made.

This is just a massive false statement. If you can't provide anything concrete on this, this discussion is over! i don't have the time or energy to continue doing this here.

dw-0, 2025-01-02:
> Any statements contrary to that are false

@Kaelum So what you're basically saying is, that a creality employee (discord name censored) is lying, gotcha.
[image]

Rest assured that we made our experiences with Creality and how they work. But you can let yourself be dazzled by their marketing for all I care.

Kaelum, 2025-01-02:
@meteyou do as you may. I answered a question that you did not like the answer to, and you just being a bully.  You are correct, being a bully does not belong on GitHub.  Your statement above says more about you, than it does about Creality or myself.  BEST

P.S. Sonic Pad and what is on the K1 line of printers is 2 completely different things.  You should know that.

dw-0, 2025-01-02:
> P.S. Sonic Pad and what is on the K1 line of printers is 2 completely different things. You should know that.

P.S. The Youtube Video linked in the first discord message of that screenshot is the advertising of the K1 --> https://www.youtube.com/watch?v=rSuFLIId7z8

meteyou, 2025-01-02:
@Kaelum why should i be a tyrant if i defend the “truth”? I only corrected your answer because it is definitely wrong and I didn't want to leave any false statements uncommented here in my/our repo. Unfortunately, you are completely wrong in your view of Creality and Klipper. You can't give me any details about Creality's “merge will” and so far I haven't seen any and I think I'm very well informed about Klipper.

why should the SonicPad and the K1 software be any different? yes more modules (PR-TOUCH) etc have been added. but both use the same “messed up” error outputs that Creality has overwritten across Klipper (they are identical from SonicPad to K1). So from my software point of view, the K1 software is only based on the SonicPad, or has simply been extended by the functions for the K1.

in short: as long as you can't share any information about the "will to merge from Creality K1 Klipper to mainline Klipper", your statement is just a rumor, but nothing more...

Kaelum, 2025-01-02:
@dw-0 you just wasted 5 minutes of my time.  There is absolutely NOTHING in that video to support ANYTHING that you have stated.  There isn't any mention of the OS at all.

meteyou, 2025-01-02:
@Kaelum pls describe why you try to "protect" Creality so hard? do you have a profit defense? 

the link to the video was only meant as a completion for the screenshot? or do you have problems translating/reading english?

this is not meant to sound negative or be an insult. i just want to understand you and that is very difficult right now...

Kaelum, 2025-01-02:
@meteyou I have tried to be cordial, but this is now at an end.  Your statement above is very offensive, so I will be reporting this to Microsoft.  Good Day

meteyou, 2025-01-02:
@Kaelum thank you for your troll attempt here. you didn't answer any of our questions and just stole our time. Good luck with the report. I'm banning you from this repo from now on. Thank you for your time!

tkunchick, 2025-01-03:
[image]

K1 software may suck (for various and multiple reasons), but the machine is decent for the price point.

sorry someone had to troll over something that seems obvious to most people.

meteyou, 2025-01-03:
i completely agree with you. the k1 should have been tested internally for 2-3 weeks longer (the initial problems with the extruder, which were fixed after a few weeks, could be fixed before the release) and the software could have been cleaned up (especially the hardcoded macros bother me the most). then this printer could really have become a very cheap and good alternative to the bambu printers. (because it "failed" of such small things, it probably annoys me even more about this product and about the customers who seek help on our discord almost daily...)

matrox471, 2025-01-26:
This issue also affect the Ender 3 V3 KE. The same error as @Atrox101 shows up. Looks like the klipper version of that printer is from the stone age of 2020 or something which is 🙄...
Also for some reasons, if i use the "Klipper Control" -> "Firmware restart" and wait on the same page, as long as i don't refresh, i have access to the toolhead controls and they works.
Made a quick video to display the behavior, dunno if it can help.

https://github.com/user-attachments/assets/d60b3574-acce-48f8-9180-62f9a51b030d

Mainsail version: v2.13.2
Moonraker version: v0.9.3.16
Printer FW version: 1.1.0.15

meteyou, 2025-01-26:
@Atrox101 thx for your post, but this PR is not merged until now and not released until now.
This PR should also fix your issue.

Valex76x, 2025-02-09:
Having the same issues on my K1 Max, is this something that will be fixed on the next Mainsail release? Any ETA available? Fot now I'm using the Creality interface just to move the toohead around 😅.

closed 2025-02-23 by meteyou

bill-orange, 2025-04-16:
I can confirm this issues still exists on latest Mainsail and latest Ender3 V3 KE firmware.  Motion Control is missing.  Restarting Klipper temporarly brings it back.

PR #2102 (2025-02-23): fix(z-tilt): Fix z_tilt check for older Klipper versions
## Description

This PR fix the z_tilt check and add a fallback for older Klipper versions in the control mixin.

## Related Tickets & Documents

fixes #2100 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none