mainsail-crew/mainsail #2101: missing the line "Overwrite action button"

AndylgTom, 2024-12-28:
### Mainsail Version:

v2.13.2

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

The localization is missing the line "Overwrite action button" for full translation.

[image]

### What did you expect to happen instead?

1

### How to reproduce this bug?

2

### Additional information:

3

meteyou, 2024-12-28:
Hi @AndylgTom ,

Thank you for your feedback! Please note that translations in Mainsail are contributed by the community, particularly by native speakers of each language. If you notice a missing or incomplete translation, you’re welcome to contribute or adjust it yourself.

You can find a guide on how to submit localization updates here: [Localization Docs](https://docs.mainsail.xyz/about/development/localization).

We appreciate any contributions that help improve Mainsail!

Best regards,
meteyou

AndylgTom, 2024-12-29:
@meteyou I mean this line is missing in all languages, including English.

meteyou, 2024-12-29:
@AndylgTom ohh sorry! I miss interpreated your bug report! I created a PR to fix this issue!

PR #2104 (2025-01-03): fix(Locale): Add missing translation key in SettingsControlTab
## Description

This PR adds a missing translation key in the SettingsControlTab for "Overwrite action button".

## Related Tickets & Documents

fixes #2101 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2025-01-03 by meteyou