mainsail-crew/mainsail #2113: Add option to update label/text of a macro prompt dialog

3b, 2025-01-11:
### Requested feature:

Add `action:prompt_update_text` and `action:prompt_update_label` or similar, that could be called between `action:prompt_show` and `action:prompt_end` to change state of currently displayed dialog.

### Solves the following problem:

While writing macros that involve making incremental changes to state (like moving toolhead) or waiting for a while, it would be nice to be able to update the text of the dialog to show things like current position  or remaining time without completely recreating the entire menu every time. Also allows factoring out the macros better, so the part doing the update doesn't need to know about the entire macro.

### Additional information:

`prompt_update_text` would probably need some way to indicate which text to update if `prompt_text` was called multiple times, either just passing an index or adding some way to name the `prompt_text`.

3b, 2025-01-11:
Possibly would be more reasonable to just recreate the dialog for every update if #2114 and #2115 were fixed, though would still allow for more flexibility in factoring the macros into pieces.

meteyou, 2025-01-11:
A command like this would not work, because it just render all commands between prompt_begin and prompt_show. So it would not read your command "update". It would also make it a lot of slower for rendering, because mainsail have to check every time the complete history, instead of the small part between "begin" and "show". so it have more negative values instead of positive possibilities.

so pls just rerender your new dialog with the new text.

closed 2025-01-11 by meteyou