mainsail-crew/mainsail #2114: Macro prompt dialogs open very slowly when there is a console on the page

3b, 2025-01-11:
### Mainsail Version:

v2.13.2

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

On screens with a console (`/console`, or `/` on desktop with a console panel), opening dialogs is very slow.
Other windows open at same time (`/config`, `/` on mobile without console panel, etc) show the dialog almost instantly.

### What did you expect to happen instead?

Dialog opens quickly regardless of screen.

### How to reproduce this bug?

Create a gcode macro that uses prompt dialogs.
```
[gcode_macro menu_test]
gcode:
  RESPOND TYPE=command MSG="action:prompt_begin start"
  RESPOND TYPE=command MSG="action:prompt_text 123"
  RESPOND TYPE=command MSG="action:prompt_text 456"
  RESPOND TYPE=command MSG="action:prompt_button 1"
  RESPOND TYPE=command MSG="action:prompt_button 2"
  RESPOND TYPE=command MSG="action:prompt_button 3"
  RESPOND TYPE=command MSG="action:prompt_button 4"
  RESPOND TYPE=command MSG="action:prompt_footer_button done|RESPOND TYPE=command MSG=action:prompt_end|warning"
  RESPOND TYPE=command MSG="action:prompt_show"

```
 Open pages with and without consoles, then activate the macro. Windows showing pages with consoles take a few seconds to show the dialog, while it is quick on others.

### Additional information:

also applies to at least firefox on android, and edge on windows.

The actual problem/cause might be "console is slow", not sure, but either way it makes debugging macros tedious.

meteyou, 2025-01-11:
it should not make any diffrenece if you have a console or not. its maybe just the rendering time of the console itself (maybe you have a huge number of new lines in your console).

so its just the GUI rendering time. nothing special to it. macro prompts are very slow an not made for "high speed inputs"

closed 2025-01-11 by meteyou

3b, 2025-01-11:
It seems to be the total number of lines in console. with empty console, it can add about 50 lines per second. With 500 lines in console it is about 2-3 per second (meaning about 3-4 seconds before a 10 line menu shows up).

Seems to drop off at around 500 rather than slowing proportional to size.

meteyou, 2025-01-11:
yeah, thats sounds reasonable! i think its more a "console bug" instead of a "macro prompt" bug