mainsail-crew/mainsail #2117: motion-project Adaptive or MJPG Streamer support

Apprisco, 2025-01-15:
### Requested feature:

Crowsnest does not seem to like running on my intel optiplex with camera-streamer for improved, stable FPS. As such, I setup a motion-project webcam which streams and snapshots properly, the website for the snapshot simply displays the image.

however, I can't seem to get it working as a webcam- it just claims error while connecting.

### Solves the following problem:

Allows more flexible webcam with FFMPEG support, simply take .jpg files from provided URL and use them as snapshots

### Additional information:

Just adding a webcam option that allows you to take .jpg or .png snapshots from a link and specify target FPS should work. Adaptive MJPG doesn't work for some reason when it's meant to do *exactly* this.
Motion-project is a well known camera streaming option.

closed 2025-01-15 by github-actions[bot]

meteyou, 2025-01-15:
adaptive is exactly what you are searching for. pls use our discord server for help.

Apprisco, 2025-01-15:
> adaptive is exactly what you are searching for. pls use our discord server for help.

I did, no answers so far. But are you sure? I've tried everything related to this and it does not work