mainsail-crew/mainsail #2123: 3D viewer completely freaks out if arc commands (G2/G3) are enabled

NameOfTheDragon, 2025-01-22:
### Mainsail Version:

2.13.2

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Suffice it to say, the 3D view cannot handle arcs:

[image]

That should look like this:

[image]

It's almost as if it's drawing the anti-arc; the part that's not supposed to be there.

### What did you expect to happen instead?

It really should support all the gcode commands including arcs G2/G3

### How to reproduce this bug?

Use Orca Slicer to slice a plate while "Arc Fitting" is enabled (this puts G2/G3 commands into the gcode).

### Additional information:

I used Orca Slicer 2.2.0 to generate my gcode.

meteyou, 2025-01-23:
pls upload a gcode file for testing

kmuuk, 2025-01-24:
Hi, I have been experiencing the same issue for some time. I also use OrcaSlicer 2.2.0 so that seems to be a commonality. 

Here is a sample gcode file: [g2g3-test.gcode.txt](https://github.com/user-attachments/files/18540785/g2g3-test.gcode.txt)

The preview is similarly messed up:
[image]

Double-checking with PrusaSlicer and an online previewer shows everything in order.
[image]

Manually stepping through the moves seems to confirm like the issuer suggests, that the inverse of the arc is drawn or that clockwise and counter-clockwise are interchanged when drawing the preview.

meteyou, 2025-01-26:
the gcode viewer is an external lib that we use. i have also seen that we are using the latest version of it. i have written to the maintainer of it and asked if he is aware of this issue.

Sindarius, 2025-01-26:
@NameOfTheDragon  Thank you for reporting this. The latest version did have a bug where the G2/G3 was not being passed to the code properly so everything was treated as a G2 which caused the arcs to render in the wrong direction. My test site was not up to date with the latest version when you checked it, can be verified by looking at the console. I have applied a fix and let @meteyou know it is ready. Here is a screenshot from my server with your sample using the new version.

Thanks again!

[image]

meteyou, 2025-01-26:
thx @Sindarius ! I updated the dependencies and now the gcode looks right.

PR #2127 (2025-01-27): fix: update Gcode-Viewer lib from sindarius to fix G2/G3 visualisation
## Description

This PR update the gcode-viewer lib to fix G2/G3 visualisation.

## Related Tickets & Documents

fixes: #2123 

## Mobile & Desktop Screenshots/Recordings

[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2025-01-27 by meteyou