mainsail-crew/mainsail #2125: Dashboard nolonger 3 columns

EyeHearYa, 2025-01-23:
### Mainsail Version:

2.13.2

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I am no longer able to have 3 columns on my dashboard after updating. When I go into settings it is set to desktop. If I click on widescreen, then close the settings window, nothing changes, reboot and still no change back to 3 columns.

### What did you expect to happen instead?

I expected to have my dashboard arranged with 3 columns like it was before I updated.

### How to reproduce this bug?

no idea

EyeHearYa, 2025-01-23:
Forgot to mention I am using orcaslicer under the device tab, for the "browser"

meteyou, 2025-01-23:
The columns are calculated by the available space. so if you browser (in your case slicer window) is too small in the width, it will just display 2 columns. you cannot change any setting to this in the settings. you can only activate/deactivate some panels in some viewports.

so just make your window bigger or check if you use the browser zoom. this can also "manipulate" the viewport. browser zoom can be managed with `ctrl` and `+ / -`.

if you need further help, pls use our discord server or the discussions section for help.