mainsail-crew/mainsail #2126: Revert to previous versions in update manager

EyeHearYa, 2025-01-23:
### Requested feature:

Please forgive my ignorance if there is already a means to achieve this, as I am still learning quite a bit about everything involved with with 3d printers, slicer software, raspberry pi, mainsail, klipper, etc. I am not sure if mainsail is even the correct focus for this concern.

I appreciate the ease of updating crowsnest, klipper, mainsail, moonraker, sonar, and the os-packages. But, if there are problems after an update, I don't see how one can easily revert to a previous version if there are any problems.

Relying on this setup for producing prototype parts and not having a means to cover my butt when things go wrong makes me nervous every time I see updates are waiting. I have just been backing up the entire hard drive at this point, which is not ideal.

Thanks for your time and attention.

agravelot, 2025-01-24:
You are gonna need to ssh to your host and checkout the specific commit you want.
https://www.git-tower.com/learn/git/faq/git-checkout-commits

meteyou, 2025-01-25:
This function itself exists in the Moonraker update manager, but there is no button in Mainsail until now. I haven't time until now to add it in the interface: https://moonraker.readthedocs.io/en/latest/web_api/#rollback-to-the-previous-version