mainsail-crew/mainsail #2130: Store Spoolman Filament ID

bofh69, 2025-01-30:
### Requested feature:

Be able to call some gcode macro or set a variable in Klipper with the filament's id when choosing a spool in mainsail.

An even more general solution would store all fields from the chosen spool's, filament's and manufacturer's in variables.

### Solves the following problem:

I've integrated Spoolman into my workflow in a different way than most.

My filament's config in the slicers have a "filament_start_gcode" setting that contains `ASSERT_ACTIVE_FILAMENT Id={{filament_id}}`.

`ASSERT_ACTIVE_FILAMENT` will see if the right filament is loaded, and if not, pause the print and tell me to switch spool. To me this makes more sense as it is the filament, not the spool, that is important for the gcode file.

If I do a multi-color print and one of the spools run out of filament, I can load the next spool of the same kind and the assert macro will still accept it at the next time the filament is switched in the gcode file.

This normally works well for me because I have a program that updates the spool & filament ids via moonraker when I switch spools. However, when that doesn't work, I would like to be able to just switch spool in mainsail and it should continue to work as usual without later pausing and complaining about the wrong filament.

I think this workflow would work better for most people. I use a program to create my filament config files from Spoolman's data, but even if someone adds the filaments themselves in their slicers, this makes it easy for them to add the macro and then get automatic verification that they have the right filament loaded when printing.

meteyou, 2025-01-30:
thx for your request, but i think this have to be more a spoolman/moonraker feature, to be able to set a filament instead of a spool. after this is implemented in spoolman/moonraker, then we could implement it in Mainsail also.

closed 2025-01-30 by meteyou

bofh69, 2025-01-30:
@meteyou I don't know Vue, mainsail's architecture and the details of the integration between frontends and moonraker so I might misunderstand things here.

From what I can see in the documentation for Moonraker, its Spoolman integration for frontends provides a proxy to spoolman's APIs that just transfers the requests and responses back and forth between Spoolman and the frontend. When mainsail query for active spools, like refreshSpools seems to be doing, mainsail should already get all the data about the spools in the response. That includes the filament's id (the filament field contains an object with an id field).

Since mainsail already shows which gcode macros exists and one can send gcode to klipper through it, I don't see what functionality you are missing in klipper nor moonraker in order to implement this feature.

An initial version can just look for a specific macro, and if it exists, call it with the `filament.id`'s value. By having a good naming scheme for the macro, it can be easy to support more fields in the future. Ie  "_MAINSAIL_SPOOLMAN_SET_filament_id" could be used.

With such an implementation, the users can choose to get the data by creating that macro. For users that don't need this, they simply don't create such a macro and everything works as usual.

meteyou, 2025-02-01:
@bofh69 Mainsail is purely a web interface and does not handle control logic. Integrating such a feature in Mainsail would mean that only users wich use Mainsail only would benefit from it, while those using other UIs like Klipperscreen, Mobilraker, or Mooncord beside Mainsail wouldn’t have issues, because the wrong filament_id would be store in this variable.

Instead, I believe the better approach would be to add a function in Moonraker to provide the filament_id (or other relevant fields) directly to Klipper. That way, all macros would have access to it without any "workaround".

If this sounds reasonable to you, the next step would be to request this feature from Moonraker/Spoolman/Klipper so that the information is always available and can be used by any macro/plugin or something else.

bofh69, 2025-02-03:
@meteyou Yes, that are valid points. I made a moonraker agent that does this instead.

meteyou, 2025-02-03:
@bofh69 thats sounds good.

would it be also possible to add a moonraker command (like the set spool command) to grab the filament_id? so it would be possible without any agent or something like that?